JChristensen/DS3232RTC #6: Disable alarms

cameraready, 2015-01-13:
I'm using your library with a chronodot v2.1 that has the DS3231 RTC. I hope to use the alarm functions that you implemented but I'm wondering how to disable an alarm after it has been set. In the tests I've run the alarms stay active after they have been set and I'm a bit confused on how to clear them. I've read the Maxim data sheet and it seems that alarms are intended be set and never changed? I don't plan on using the interrupt for the alarms but have been checking them with the alarm(ALARM_1) function. Would the alarmInterrupt(ALARM_1, false) function disable the alarm even if not using interrupts?

Thanks,

Ross

JChristensen, 2015-01-13:
Hi Ross,

Alarms can be set and changed any number of times, there is no limit. Use the `setAlarm()` function.

`alarmInterrupt(ALARM_n, false)` disables the external alarm signal on the INT pin; however, the alarm functionality continues to operate internally and will set the RTC's internal alarm flag when a match condition occurs, whether the external alarm signal is enabled or not. I'm not aware that this functionality can be disabled.

The `alarm(ALARM_n)` function reads the internal flag, clears it, then returns the value that it read.

If the external alarm signal is not being used, then it's the responsibility of the application firmware (i.e. the Arduino sketch) to maintain an ALARM ON/OFF state if so desired. In short, if the alarm is off, then either don't call `alarm()` or disregard its return value.

Hope this helps.  If not clear, let me know.

Jack

Closes #6

closed 2015-01-13 by JChristensen

cameraready, 2015-01-13:
That makes sense. I'll setup some local variables to chose whether to act on the alarms in the sketch.

Thanks,

Ross