JChristensen/DS3232RTC #7: Power ON/Off Microcontroller with alarmInterrupt()

totalsoft-petrache, 2015-02-05:
Hello,
I'm trying to create a project with RTC3231 and Arduino, in which I set an alarmInterrupt() which wakes up the Arduino and does a specific task at RTC.alarm(ALARM_1). 

My problem is how do I connect INT/SQW to PIn 2 on Arduino? I'm not sure if I need a resistor or MOSFET (as found on a site) for when the alarmInterrupt triggers INT pin to go LOW no to fry the Arduino.
Also, you think that putting arduino to sleep for 2 days will be dangerous?

Thanks

 ```
/*----------------------------------------------------------------------*
 * Demonstrates usage of the alarm interrupt for alarm 1 and alarm 2.   *
 *                                                                      *
 * 26 December 2014.                                                    *
 *                                                                      *
 * Tested with Arduino 1.5.8, Arduino Nano w/ ATmega328, DS3231.        *
 *                                                                      *
 * This work is licensed under the Creative Commons Attribution-        *
 * ShareAlike 3.0 Unported License. To view a copy of this license,     *
 * visit http://creativecommons.org/licenses/by-sa/3.0/ or send a       *
 * letter to Creative Commons, 171 Second Street, Suite 300,            *
 * San Francisco, California, 94105, USA.                               *
 *----------------------------------------------------------------------*/ 

#include <DS3232RTC.h>        //http://github.com/JChristensen/DS3232RTC
#include <Streaming.h>        //http://arduiniana.org/libraries/streaming/
#include <Time.h>             //http://playground.arduino.cc/Code/Time
#include <Wire.h>             //http://arduino.cc/en/Reference/Wire

#define SQW_PIN 2

void setup(void)
{
    Serial.begin(115200);

    //setSyncProvider() causes the Time library to synchronize with the
    //external RTC by calling RTC.get() every five minutes by default.
    setSyncProvider(RTC.get);
    Serial << "RTC Sync";
    if (timeStatus() != timeSet){
        Serial << " FAIL!";
    }
    Serial << endl;

    printDateTime( RTC.get() );
    Serial << " --> Current RTC time." << endl;

    //Disable the default square wave of the SQW pin.
    RTC.squareWave(SQWAVE_NONE);

    //Attach an interrupt on the falling of the SQW pin.
    //digitalWrite(SQW_PIN, HIGH);    //redundant with the following line
    pinMode(SQW_PIN, INPUT_PULLUP);
    attachInterrupt(INT0, alarmIsr, FALLING);

    //Set an alarm at every 20th second of every minute.
    RTC.setAlarm(ALM1_MATCH_SECONDS, 20, 0, 0, 1);    //daydate parameter should be between 1 and 7
    RTC.alarm(ALARM_1);                   //ensure RTC interrupt flag is cleared
    RTC.alarmInterrupt(ALARM_1, true);

    //Set an alarm every minute.
    RTC.setAlarm(ALM2_EVERY_MINUTE, 0, 0, 0, 1);    //daydate parameter should be between 1 and 7
    RTC.alarm(ALARM_2);                   //ensure RTC interrupt flag is cleared
    RTC.alarmInterrupt(ALARM_2, true);
}

volatile boolean alarmIsrWasCalled = false;

void alarmIsr()
{
    alarmIsrWasCalled = true;
}

void loop(void)
{
    if (alarmIsrWasCalled){
        if (RTC.alarm(ALARM_1)) {
            printDateTime( RTC.get() );
            Serial << " --> Alarm 1!" << endl;
        }
        if (RTC.alarm(ALARM_2)) {
            printDateTime( RTC.get() );
            Serial << " --> Alarm 2!" << endl;
        }
        alarmIsrWasCalled = false;
    }
}

void printDateTime(time_t t)
{
    Serial << ((day(t)<10) ? "0" : "") << _DEC(day(t)) << ' ';
    Serial << monthShortStr(month(t)) << " " << _DEC(year(t)) << ' ';
    Serial << ((hour(t)<10) ? "0" : "") << _DEC(hour(t)) << ':';
    Serial << ((minute(t)<10) ? "0" : "") << _DEC(minute(t)) << ':';
    Serial << ((second(t)<10) ? "0" : "") << _DEC(second(t));
}
```

JChristensen, 2015-02-05:
The INT/SQW pin is a low-power signal, it can simply be connected directly to pin 2 on the microcontroller. However, it is an open-drain pin so a pullup resistor is required. This can be an external resistor, but I would prefer to use the microcontroller's internal pullup resistor on pin 2.

There is no danger in having the microcontroller sleep for any arbitrarily long period.

You should be aware that most Arduino boards have voltage regulators, LEDs and other support components that draw power continuously, and that this greatly diminishes the power savings to be had by sleeping the microcontroller. Either modifying the board or using a custom design will be required to realize maximum power savings. [Here](http://forum.arduino.cc/index.php?topic=164146.msg1225643) is one discussion on the Arduino forum; there are many others, and many other sources on the web. As always, GIYF.

closed 2015-02-05 by JChristensen

totalsoft-petrache, 2015-02-05:
Thanks for the answer. I saw the discussion after I posted the question. 

Unfortunately now I have a compilation problem with the code above:

In file included from sketch_feb05a.ino:15:
C:\Program Files (x86)\Arduino\libraries\DS3232RTC-master/DS3232RTC.h:132: error: 'time_t' does not name a type
C:\Program Files (x86)\Arduino\libraries\DS3232RTC-master/DS3232RTC.h:133: error: 'time_t' has not been declared
C:\Program Files (x86)\Arduino\libraries\DS3232RTC-master/DS3232RTC.h:134: error: 'tmElements_t' has not been declared
C:\Program Files (x86)\Arduino\libraries\DS3232RTC-master/DS3232RTC.h:135: error: 'tmElements_t' has not been declared
sketch_feb05a:26: error: variable or field 'printDateTime' declared void
sketch_feb05a:26: error: 'time_t' was not declared in this scope
sketch_feb05a.ino: In function 'void setup()':
sketch_feb05a:28: error: 'class DS3232RTC' has no member named 'get'
sketch_feb05a:28: error: 'setSyncProvider' was not declared in this scope
sketch_feb05a:29: error: no match for 'operator<<' in 'Serial << "RTC Sync"'
sketch_feb05a:30: error: 'timeStatus' was not declared in this scope
sketch_feb05a:30: error: 'timeSet' was not declared in this scope
sketch_feb05a:31: error: no match for 'operator<<' in 'Serial << " FAIL!"'
sketch_feb05a:33: error: 'endl' was not declared in this scope
sketch_feb05a:35: error: 'class DS3232RTC' has no member named 'get'
sketch_feb05a:35: error: 'printDateTime' was not declared in this scope
sketch_feb05a:36: error: no match for 'operator<<' in 'Serial << " --> Current RTC time."'
sketch_feb05a.ino: In function 'void loop()':
sketch_feb05a:68: error: 'class DS3232RTC' has no member named 'get'
sketch_feb05a:68: error: 'printDateTime' was not declared in this scope
sketch_feb05a:69: error: no match for 'operator<<' in 'Serial << " --> Alarm 1!"'
sketch_feb05a:69: error: 'endl' was not declared in this scope
sketch_feb05a:72: error: 'class DS3232RTC' has no member named 'get'
sketch_feb05a:72: error: 'printDateTime' was not declared in this scope
sketch_feb05a:73: error: no match for 'operator<<' in 'Serial << " --> Alarm 2!"'
sketch_feb05a:73: error: 'endl' was not declared in this scope
sketch_feb05a.ino: At global scope:
sketch_feb05a:79: error: variable or field 'printDateTime' declared void
sketch_feb05a:79: error: 'time_t' was not declared in this scope

JChristensen, 2015-02-05:
Reread the installation section in the [ReadMe file](https://github.com/JChristensen/DS3232RTC/blob/master/ReadMe.md), especially the part about renaming the downloaded library. Also ensure that the Streaming and Time libraries are installed and in the correct directory. Finally, note that `arduino\sketchbook\libraries` is the preferred directory in which to install libraries.