JChristensen/DS3232RTC #8: read alarm times, control registers

kunaakos, 2015-03-04:
I'm using a module based on this chipset for an alarm clock, and the problem is that I cannot read settings from the module with the library. Things like the times/dates the alarms are set to, control registers. These could be stored in flash (but the DS3231 that I have has no flash) or EEPROM, but that's just a bunch of extra code, and gives room for errors.
I forked your library, added methods to read alarm times, used the methods for reading time values as a base:
<code>time_t getAlarm(byte alarmNumber);</code>
<code>byte readAlarm(byte alarmNumber, tmElements_t &tm);</code>
Would you be interested in adding this, should I submit a pull request? Works for me so far, I test what I can, but I wouldn't trust myself :grin:
I'm also thinking of adding a method to get the alarm interrupt enable (A1IE, A2IE) settings, because I'll use these to turn alarms on or off in my project.

ChuckBland, 2015-03-04:
> Works for me so far, I test what I can, but I wouldn't trust myself [image:

Nice....

JChristensen, 2015-03-04:
Thanks for the note, glad you were able to use the library and modify it for your own requirements. I appreciate the offer; however one main goal I had for this library was to keep it lightweight with a small code footprint. `readRTC()` and `writeRTC()` can be used to manipulate any of the RTC's registers; of course this includes the alarm registers. Symbolic names for all registers are defined in the .h file. Admittedly this may not be quite as convenient as a dedicated method. I actually did consider including functions to read the alarm registers but decided in favor of a smaller library. Further, I couldn't think of a lot of scenarios aside from debugging where I would really need to read them. If I had some reason to question what was in the registers, it seems like I'd be just as well off ignoring the current contents and overwriting them with the desired values.

closed 2015-03-04 by JChristensen

kunaakos, 2015-03-04:
Whoa there, I thought <code>readRTC()</code> and <code>writeRTC()</code> were for the extra SRAM found on the DS3232 only. :disappointed: Well, it was a great learning experience. Thanks for clearing things up!
Regarding reading the registers: after an accidental restart, blackout etc. it might be cool to not overwrite the alarm settings, and at that point you don't even know if there's anything in there at all if you can't read. A random alarm might be triggered if you don't overwrite, you might erase an important alarm if you do.

JChristensen, 2015-03-05:
All experiences are good ones :smile:  Check the code for functions like `squareWave()`, `oscStopped()` or `temperature()`. They just call `readRTC()` and `writeRTC()` to manipulate the appropriate registers.

Note that there is *always* something in the alarm registers and that there is no way to actually turn the alarm functionality off! It's always comparing the alarm registers to the current time and will set the alarm flags A1F and A2F in the status register when appropriate matches occur. If alarms are not required, then they can just be ignored.

The INTCN, A1IE and A2IE bits in the control register determine whether an "interrupt" (alarm) signal is output on the INT/SQW pin.

Finally, consider that if the RTC has a backup battery, then a power-on or reset condition for the microcontroller does not reset the RTC. After a reset, the MCU would normally initialize the RTC as appropriate for the application. Reading the alarm registers at that point would require that the MCU know what *should* be in them in order to make a decision on what to do. I suppose this is a possibility though, depending again on the application.

kunaakos, 2015-03-05:
Thanks again for your help, I checked other issues, and learned from your other replies too, much appreciated. :beers:

Turning the alarm "off": I figure I have to possibilities, either set alarms that match the next day only so it won't trigger again after that (which would be messy, since my clock is simple and won't display dates, so I only want to work with time values), or use the A1IE and A2IE registers as alarm status flags. Interrupt enabled means that the alarm is on, disabled is off. Works even if I don't use interrupts...

After a reset, I read ALL alarm settings from the RTC (alarm type etc), my code doesn't even store these, I just read everything on the fly when checking an alarm - this is not completely implemented yet, reading and setting alarms works, but I cannot turn off an alarm yet, so theory only.

In the real world this would not be ideal (bunch of I2C reads drain battery and slow things down), but this is not a battery-operated project, and I'm aiming for writing my code in decoupled modules (a class in separate .h and .cpp files) - every piece of hardware used (RTC, sound module, phone dial etc. in my project) has it's own module.

If I replace hardware, I just drop in another module without having to rewrite, modules are reusable etc. Similar to libraries, but this provides an extra layer between the libs and my sketch. All configuration goes inside modules, so replacing a module means that you only have to swap two files. Module might not even be the proper term for what I'm doing here...

I've been dabbling in web and MVC, so I'm kind of trying to used what I learned there in Arduino. Again, I do know that this can increase compiled size, so in the real world... :grin: Didn't completely figure out how to put all of this together, since I don't want to use any other loop or any delays within the main loop and modules can't communicate with each other, but that's what today is for.