JChristensen/DS3232RTC #10: Alarm not setting

JArduinoNew, 2015-05-01:
I am working on a code that uses RTC chronoDot to wake up an arduino at regular iterval.  In the debugging process right now. Using JChristense DS3232RTC library.  The problem is that for some reason the alarm is not being triggered (in fact I don't even know if its being set)  I have tried several alarm options to no avail.  Any help will be greatly appreciated.

I enclose a snippet of the code I am using.

#include <DS3232RTC.h>    //http://github.com/JChristensen/DS3232RTC
#include <Time.h>         //http://www.arduino.cc/playground/Code/Time
#include <Wire.h>         //http://arduino.cc/en/Reference/Wire (included with Arduino IDE)

#define setAlarmType2 ALM2_MATCH_HOURS
#define setAlarmType1 ALM1_MATCH_MINUTES
#define SQW_PIN 2

time_t myTime;
tmElements_t tm;
float yearSince=0.0;
int currentYear=0;
int partYear=0;
volatile int gpsCounter=0;
int interruptPIN = 10;

void setup() {
  Serial.begin(57600);
  pinMode(interruptPIN, OUTPUT);
  RTC.squareWave(SQWAVE_NONE);
  pinMode(SQW_PIN, INPUT_PULLUP);
  attachInterrupt(INT0, wakeFxn, FALLING);      //Could just say 0, ISR, LOW

}

void loop() {
  myTime=RTC.get();
  //yearSince=((myTime)/ 31556926);
  //partYear =(myTime%31556926);
  //currentYear=1970+yearSince;
  Serial.println(myTime);
  RTC.read(tm);
  Serial.print(tm.Hour,DEC);
  Serial.print(':');
  Serial.print(tm.Minute,DEC);
  Serial.print(':');
  Serial.println(tm.Second,DEC);
  Serial.print(tm.Month, DEC);
  Serial.print('/');
  Serial.print(tm.Day,DEC);
  Serial.print('/');
  Serial.println(tm.Year+1970);
  //Serial.println(currentYear);
  //Serial.println(yearSince);
  //Serial.println(partYear);
  delay(5000);
  RTC.setAlarm(ALM1_MATCH_SECONDS, 20, 0, 0, 1);
  RTC.setAlarm(ALM1_MATCH_DAY, 00, 20, 14, dowFriday);
  RTC.alarm(ALARM_1);
  RTC.alarmInterrupt(ALARM_1, true);
  RTC.setAlarm(setAlarmType2, 00, 14, 15);          //Alarm 2 once a day
  RTC.alarmInterrupt(ALARM_2, true);
  
  if (RTC.alarm(ALARM_1)){
    Serial.println("Alarm 1 was triggered");
    digitalWrite(interruptPIN, HIGH);
    delay(10000);
  }
  
  if (RTC.alarm(ALARM_2)){
    Serial.println("Alarm 2 was triggered");
  }
}

void wakeFxn(){}

JArduinoNew, 2015-05-01:
I realized I had a line RTC.alarm() after setAlarm that was resetting the alarm flag.  What do you think would be the best way to set the alarm every 3 hours?

Thank you

JChristensen, 2015-05-02:
Please raise issues here only if there is a demonstrable problem with the library; this is not the place to debug your code. Suggest posting on the Arduino forum instead.

closed 2015-05-02 by JChristensen