JChristensen/DS3232RTC #14: ALM1_MATCH_DAY sets wrong alarm type

pchandle, 2015-09-24:
Setting an alarm of type ALM1_MATCH_DAY  doesn't set the Alarm 1 DY/nDT bit. I believe this is because the  code in the setAlarm routine that reads:
   if (alarmType & 0x10) hours |= _BV(DYDT);
should be:
   if (alarmType & 0x10) daydate |= _BV(DYDT);

That is, bit 6 of the wrong register is being set.

Apollon77, 2015-09-24:
I think you are right ... According to https://datasheets.maximintegrated.com/en/ds/DS3232.pdf the bit 6 in "hours" is to set 12/24h-clock...
And it should have effect for ALM2 too, or !?

pchandle, 2015-09-24:
Yes the change applies to ALM2 as well. so the proposed code change will work/apply to ALM1 and ALM2.

JChristensen, 2018-01-01:
Fixed by https://github.com/JChristensen/DS3232RTC/commit/d004bb286c19d0577d13e2f2c513b12aab7b9b0e

closed 2018-01-01 by JChristensen