JChristensen/DS3232RTC #18: Enabling BBSQW mode on RTC DS3231

superbaj1986, 2015-11-20:
Hi jack,
    I need to enable the battery backup sqarewave interruption to wakeup arduino, the interruption working fine when connected to the vcc pin but not on battery running, I don't know why, in one of your thread I find your below said line to enable the battery RTC interrupt but I don't know where to put that code and how can you put give me an idea about and also I need to check whether this register is set it not, please help me.

RTC.writeRTC(RTC_CONTROL, (RTC.readRTC(RTC_CONTROL) | _BV(BBSQW)));

martin-eden, 2020-01-10:
You can place it in `DS3231RTC.cpp:begin()`:
```C
// Initialize the I2C bus.
void DS3232RTC::begin()
{
    i2cBegin();
    RTC.writeRTC(RTC_CONTROL, (RTC.readRTC(RTC_CONTROL) | _BV(BBSQW)));
}
```

roblatour, 2021-04-29:
I tried that but am getting this error when I try to compile:

C:\Users\xxx\Documents\Arduino\libraries\DS3232RTC\src\DS3232RTC.cpp:127:3: error: 'RTC' was not declared in this scope
   RTC.writeRTC(RTC_CONTROL, (RTC.readRTC(RTC_CONTROL) | _BV(BBSQW)));
   ^
exit status 1

JChristensen, 2021-04-29:
The register definitions aren't exposed to the user API, but maybe should be.
Rather than changing `DS3232RTC.cpp:begin()`, I'd recommend doing it in your sketch as follows.
```c++
#define RTC_CONTROL 0x0e
#define BBSQW 6

void setup()
{
    RTC.begin();
    RTC.writeRTC(RTC_CONTROL, (RTC.readRTC(RTC_CONTROL) | _BV(BBSQW)));
}
```

roblatour, 2021-04-29:
Perfect, thank you!

JChristensen, 2021-04-29:
What board are you using? For something other than AVR architecture, a DS3232RTC object will need to be instantiated, [see the note here](https://github.com/JChristensen/DS3232RTC#constructor).

closed 2021-05-02 by JChristensen

roblatour, 2021-05-02:
I was using an esp32 devkitc_v4 (board manager doit esp32 devkit v1) - and the recommendation works great for me.

JChristensen, 2021-05-02:
Thanks for the feedback!

roblatour, 2021-05-04:
Sorry, I got that wrong.  I had some wiring mixed up.  In fact, what happens is that immediately after I put the esp32 into deep sleep, it wakes up again.

Here is the code I am using to put it to sleep:

` int nextWakeupMinute = ( minute() + NumberOfMinutesBetweenReadings ) % 60;

  rtc.setAlarm(ALM1_MATCH_MINUTES, 0, nextWakeupMinute, 0, 0);
  rtc.alarm(ALARM_1);
  rtc.squareWave(SQWAVE_NONE);
  rtc.alarmInterrupt(ALARM_1, true);

  pinMode(interruptPin, INPUT_PULLUP);

  rtc_gpio_pullup_en(interruptPin);
  rtc_gpio_pulldown_dis(interruptPin);
  esp_sleep_enable_ext0_wakeup(interruptPin, 0);

  esp_deep_sleep_start();`

Prior to this code, the RTC is connected to power and other functions (like getting the time from it) work fine.  However, once the ESP goes into deep sleep the power is dropped.  The DS3231 does have a battery in it.  The goal is to have it wake up at the right time when only powered by the battery.  However, as stated above, what happens is the ESP32 wakes up immediately.

Here is what I see in the Serial Monitor:

`10:50:16.282 -> Go to sleep based on the RTC
10:50:16.282 -> 
10:50:16.282 -> ets Jun  8 2016 00:22:57
10:50:16.282 -> 
10:50:16.282 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
10:50:16.282 -> configsip: 0, SPIWP:0xee
10:50:16.282 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
10:50:16.282 -> mode:DIO, clock div:1
10:50:16.282 -> load:0x3fff0018,len:4
10:50:16.282 -> load:0x3fff001c,len:1044
10:50:16.282 -> load:0x40078000,len:10124
10:50:16.282 -> load:0x40080400,len:5856
10:50:16.282 -> entry 0x400806a8
`

Having that said, if I change the wiring so the DS3231 is powered continuously, ie power does not drop out when the esp32 goes to sleep, then everything works fine - that is to say, the esp wakes up only at the prescribed time.

JChristensen, 2021-05-04:
This sounds like an issue with the ESP32 code as opposed to an issue with the library. I don't use any of the ESP boards so I find it hard to comment. I have used the library to wake AVR microcontrollers in various projects, [here is one](https://github.com/JChristensen/aaXBee), FWIW.

JChristensen, 2021-05-04:
Or it could be a hardware issue, given your last sentence above.