JChristensen/DS3232RTC #19: Interrupts from DS3231

davidsteinsland, 2016-01-06:
Per the documentation, shouldn't the SQW pin go low when an alarm interrupt occurs? Per your API, I've set up repeated alarms like so:

```c
RTC.setAlarm(ALM2_EVERY_MINUTE, 0, 0, 0, 0);
RTC.alarmInterrupt(ALARM_2, true);
```

This works well by just checking `RTC.alarm(ALARM_2)` in the `loop`, but this is a complete waste of power. So, I hooked by multimeter to the SQW pin, but this never gets pulled low. It's constant at about 2.5 volts.

Are you sure the alarm interrupts are being written properly to the register?

t3chguy, 2016-01-06:
try removing the `RTC.alarm(ALARM_2)` from your loop, the pulse will be so short that the multimeter won't see it.

t3chguy, 2016-01-06:
also, I assume you have a pull up on the SQW pin, as its open drain.

davidsteinsland, 2016-01-06:
But that wont reset the alarm after it fires, meaning it will only fire once?
As far as I've understood one must constantly check with `RTC.alarm(ALARM_2)` to reset the status register each time the alarm goes off?

Yes, I've got a 10kΩ pullup.

t3chguy, 2016-01-06:
correct, but if you remove the reset then you can see it physically. on the Code side you should only call reset after seeing the Interrupt, basically:
Arduino watches for SQW to fall, then queries which Alarm fired.
If you purposefully don't reset the alarms in the code, then you can test whether the SQW pin will actually work with your Multimeter

davidsteinsland, 2016-01-06:
Ah, yes, believe I have it somewhat working now.
Only problem being that if I try to reset the alarm from within the interrupt/ISR routine, my arduino freezes.

Sample:
```c
void alarmChange() {
  uint8_t rs = RTC.readRTC(RTC_STATUS);
  if (rs & (1 << 0)) {
    Serial.println("Alarm 1");
  } else {
    Serial.println("Alarm 2");
  }
}
```

Although this code only queries the alarm type, it will still freeze up. This means that I will have to have a `loop` like this:
```c
void loop() {
  if (RTC.alarm(ALARM_1)) {
  }
  
  if (RTC.alarm(ALARM_2)) {
  }
  
  delay(500);
}
```
which is the thing I'm trying to avoid (I don't want to have a loop function at all).

t3chguy, 2016-01-06:
You should never do anything intensive in an ISR, set a volatile variable and have the loop check it

t3chguy, 2016-01-06:
Example: https://github.com/JChristensen/DS3232RTC/issues/5#issuecomment-68143652

davidsteinsland, 2016-01-06:
That's just it – I'm building a device that should only be active for 1 or 2 times a day, and in order to save power, I don't want to have a constant pull-for-something in the `loop` (I want my arduino to go to sleep)

t3chguy, 2016-01-06:
Well find a way of having an intensive routine in an ISR, you'll probably have to go beyond what you can do simply from the IDE, You can in theory do intensive subroutines but they'll often break.

JChristensen, 2016-01-06:
@davidsteinsland, where was the RTC purchased? 2.5V sounds like an odd reading, unless the power supply is 2.5V. Wasting power is a relative thing, but I have multiple low power projects where the MCU sleeps and the RTC interrupt wakes it, so I'm confident that the alarm functionality works as advertised.
If there is a demonstrable problem with the library, please submit a [SSCCE](http://www.sscce.org/) with a description of expected vs. actual behavior.

closed 2016-01-06 by JChristensen

davidsteinsland, 2016-01-06:
@JChristensen sorry, I remembered wrong; it was 4,8V.
If your MCU is sleeping, then you aren't performing any logic in your `loop`?

Basically, I now have something like this:
```
void loop() {
  // alarmCalled is set from an ISR
  if (alarmCalled) {
    alarmCalled = 0;
    if (RTC.alarm(ALARM_1)) {
      alarm1(); 
    }
    
    if (RTC.alarm(ALARM_2)) {
      alarm2();
    }
  }
}
```

What I'm trying to do is something like:
```
[sleeping arduino] --> [receive interrupt via ISR] --> [wake up] --> [do work] --> [repeat]
```

JChristensen, 2016-01-06:
@davidsteinsland, sure, in fact all the logic is in loop() and the ISR is empty, as it only serves to wake the MCU; no actual work is done in the ISR.
FYI I would prefer to limit issues and discussion here to actual problems with the library itself as opposed to issues with projects that folks are working on. Best to take those discussions to the Arduino forum, StackExchange or whatever.
[Here](https://github.com/JChristensen/aaXBee_HW) is a project that may be instructive, and [here](https://gist.github.com/JChristensen/5616922) are some basic sleep demos. Good luck with your project!