JChristensen/DS3232RTC #20: Getter for alarms

davidsteinsland, 2016-01-07:
Would be nice if the library had an API for querying the current alarms for ALM1 and ALM2. The getter could return a struct, like:

```c
alarm_t alarm_def;
if (RTC.getAlarm(ALARM_2, &alarm_def)) {
    if (alarm_def.type == ALM2_MATCH_HOURS) {
        Serial.print("I will trigger at ");
        Serial.print(alarm_def.hour);
        Serial.print(':');
        Serial.print(alarm_def.minute);
        Serial.println(" every day.");
    }
}
```

JChristensen, 2016-01-07:
That crossed my mind but it didn't seem terribly useful. Assuming that I set the alarm(s) then it seems that there would be little need to read them back. If I were for some reason unsure of how they were set, it seems more straightforward to just set them as desired. I could probably think of a scenario where such a thing would be convenient but I don't have bandwidth for the development currently. The readRTC() function provides a lower level ability to read the alarm registers. If that's not good enough then I'd fork the repo to add the desired functionality or writing a derived class.

closed 2016-01-07 by JChristensen

davidsteinsland, 2016-01-07:
Yeah, I better read the registers myself. I'm outputting the alarm onto a LCD, allowing a user to change it. So that's the reason I wanted to read the current alarm (to know the time, but also to know if there is an alarm activated or not).

JChristensen, 2016-01-07:
I might persist a copy of the alarm settings in SRAM. Then the only time it would be necessary to read the alarms from the RTC would be after a power cycle, when SRAM contents would be lost.

davidsteinsland, 2016-01-08:
Yeah, I ended up writing this:

```c
/*
  assuming 24 hour clock.
  does not fetch day/date
 */
void readAlarm(uint8_t alarm_number, alarm_t* alarm) {
  uint8_t addr;

  if (alarm_number == ALARM_1) {
    addr = ALM1_SECONDS;
  } else {
    addr = ALM2_MINUTES;
  }

  uint8_t seconds = 0, minutes, hour, date, alarm_type = 0;
  
  if (alarm_number == ALARM_1) {
    RTC.readRTC(addr++, &seconds, 1);
  }

  RTC.readRTC(addr, &minutes, 1);
  RTC.readRTC(addr + 1, &hour, 1);
  RTC.readRTC(addr + 2, &date, 1);

  /* get the 8th bit and shift it right to 2^4 */
  alarm_type = (date & (1 << 7)) >> 4;
  /* get the 8th bit and shift it right to 2^3 */
  alarm_type = alarm_type | ((hour & (1 << 7)) >> 5);
  /* get the 8th bit and shift it right to 2^2 */
  alarm_type = alarm_type | ((minutes & (1 << 7)) >> 6);
  /* get the 8th bit */
  alarm_type = alarm_type | ((seconds & (1 << 7)) >> 7);

  if (alarm_number == ALARM_2) {
    /* shift one bit to the right, removing seconds */
    alarm_type = alarm_type >> 1;
  }

  /* does not fetch DY/DT, thus ignoring ALM2_MATCH_DAY and ALM2_MATCH_DATE */
  switch (alarm_type) {
    case 16:
      alarm_type = ALM1_MATCH_DAY;
    case 15:
      alarm_type = ALM1_EVERY_SECOND;
      break;
    case 14:
      alarm_type = ALM1_MATCH_SECONDS;
      break;
    case 12:
      alarm_type = ALM1_MATCH_MINUTES;
      break;
    case 8:
      if (alarm_number == ALARM_1) {
        alarm_type = ALM1_MATCH_HOURS;
      } else {
        alarm_type = ALM2_MATCH_DAY;
      }
      break;
    case 7:
      alarm_type = ALM2_EVERY_MINUTE;
      break;
    case 6:
      alarm_type = ALM2_MATCH_MINUTES;
      break;
    case 4:
      alarm_type = ALM2_MATCH_HOURS;
      break;
    case 0:
      if (alarm_number == ALARM_1) {
        alarm_type = ALM1_MATCH_DATE;
      } else {
        alarm_type = ALM2_MATCH_DATE;
      }
      break;
  }

  alarm->alarm = alarm_number;
  alarm->type = alarm_type;
  /* ignoring 8th bit */
  alarm->seconds = bcd2dec(seconds & ~(1 << 7));
  alarm->minutes = bcd2dec(minutes & ~(1 << 7));
  /* ignoring 8th and 7th bit */
  alarm->hours = bcd2dec(hour & ((1 << 6) - 1));
}
```

with the struct 
```c
typedef struct {
  uint8_t alarm;
  uint8_t type;
  uint8_t hours;
  uint8_t minutes;
  uint8_t seconds;
} alarm_t;
```