JChristensen/DS3232RTC #24: Several error on Arduino Zero, M0 Pro, Due

mieperDE, 2016-03-13:
Example TimeRTC
Using the Arduino Zero, M0 Pro and Arduino Due

> In file included from C:\Users\Phil\Documents\Arduino\libraries\DS3232RTC-master\examples\TimeRTC\TimeRTC.ino:8:0:
> 
> C:\Users\Phil\Documents\Arduino\libraries\DS3232RTC-master/DS3232RTC.h:136:26: error: 'tmElements_t' has not been declared
> 
>          static byte read(tmElements_t &tm);
> 
>                           ^
> 
> C:\Users\Phil\Documents\Arduino\libraries\DS3232RTC-master/DS3232RTC.h:137:20: error: 'tmElements_t' has not been declared
> 
>          byte write(tmElements_t &tm);
> 
>                     ^
> 
> In file included from C:\Users\Phil\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel/Device/ATMEL/samd21/include/samd21.h:55:0,
> 
>                  from C:\Users\Phil\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel/Device/ATMEL/samd.h:91,
> 
>                  from C:\Users\Phil\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel/Device/ATMEL/sam.h:581,
> 
>                  from C:\Users\Phil\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.4\cores\arduino/Arduino.h:47,
> 
>                  from sketch\TimeRTC.ino.cpp:1:
> 
> C:\Users\Phil\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel/Device/ATMEL/samd21/include/samd21g18a.h:486:38: error: expected ')' before '*' token
> 
>  #define RTC               ((Rtc      *)0x40001400U) /**< \brief (RTC) APB Base Address */
> 
>                                       ^
> 
> C:\Users\Phil\Documents\Arduino\libraries\DS3232RTC-master/DS3232RTC.h:155:18: note: in expansion of macro 'RTC'
> 
>  extern DS3232RTC RTC;
> 
>                   ^
> 
> C:\Users\Phil\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel/Device/ATMEL/samd21/include/samd21g18a.h:486:38: error: expected ')' before '*' token
> 
>  #define RTC               ((Rtc      *)0x40001400U) /**< \brief (RTC) APB Base Address */
> 
>                                       ^
> 
> C:\Users\Phil\Documents\Arduino\libraries\DS3232RTC-master/DS3232RTC.h:155:18: note: in expansion of macro 'RTC'
> 
>  extern DS3232RTC RTC;
> 
>                   ^
> 
> C:\Users\Phil\Documents\Arduino\libraries\DS3232RTC-master\examples\TimeRTC\TimeRTC.ino: In function 'void setup()':
> 
> TimeRTC:15: error: request for member 'get' in '1073746944u', which is of pointer type 'Rtc*' (maybe you meant to use '->' ?)
> 
>      setSyncProvider(RTC.get);   // the function to get the time from the RTC
> 
>                          ^
> 
> TimeRTC:15: error: 'setSyncProvider' was not declared in this scope
> 
>      setSyncProvider(RTC.get);   // the function to get the time from the RTC
> 
>                             ^
> 
> TimeRTC:16: error: 'timeStatus' was not declared in this scope
> 
>      if(timeStatus() != timeSet) 
> 
>                    ^
> 
> TimeRTC:16: error: 'timeSet' was not declared in this scope
> 
>      if(timeStatus() != timeSet) 
> 
>                         ^
> 
> C:\Users\Phil\Documents\Arduino\libraries\DS3232RTC-master\examples\TimeRTC\TimeRTC.ino: In function 'void digitalClockDisplay()':
> 
> TimeRTC:31: error: 'hour' was not declared in this scope
> 
>      Serial.print(hour());
> 
>                        ^
> 
> TimeRTC:32: error: 'minute' was not declared in this scope
> 
>      printDigits(minute());
> 
>                         ^
> 
> TimeRTC:33: error: 'second' was not declared in this scope
> 
>      printDigits(second());
> 
>                         ^
> 
> TimeRTC:35: error: 'day' was not declared in this scope
> 
>      Serial.print(day());
> 
>                       ^
> 
> TimeRTC:37: error: 'month' was not declared in this scope
> 
>      Serial.print(month());
> 
>                         ^
> 
> TimeRTC:39: error: 'year' was not declared in this scope
> 
>      Serial.print(year()); 
> 
>                        ^
> 
> exit status 1
> request for member 'get' in '1073746944u', which is of pointer type 'Rtc*' (maybe you meant to use '->' ?)
>

JChristensen, 2016-03-13:
This library was written for the AVR MCUs and is known not to compile for the ARM architecture. No current plans to address this, sorry.

closed 2016-03-13 by JChristensen

mieperDE, 2016-03-13:
Damn :)
Fortunately, the DS1307RTC library works with ARM

1cc386ea8e470a, 2018-01-27:
Jack,
Any plans to support ARM? Please? Pretty Please?

JChristensen, 2018-01-28:
@1cc386ea8e470a, no plans I'm afraid. I haven't gotten into the ARM chips (although they do look interesting). Which boards are you using? I can get the library to compile clean for Arduino Due or Zero with just a couple small changes, but lacking ARM hardware myself, I can't test it; there could well be logic errors to be addressed. Even if there were no logic errors, then I am in the position of fielding issues from folks with ARM hardware when I don't use that hardware myself. So that just isn't sounding like somewhere I want to go, I hardly have time to support my existing repos properly. Very sorry.

1cc386ea8e470a, 2018-01-28:
Jack,
Thanks for getting back to me! I'm using an adafruit feather which has a 
Cortex M0 chip. Same as a Zero I believe. I'd be happy to donate one to you 
for the cause if you would like!  ;)
But I get what you're saying about the ongoing support issue. Your 
libraries (particularly the timezone one) are the best around. The ARM 
chips are so much faster, but beyond that, the fact that they have a large 
amount of memory makes them so attractive to me. Anyways, if you change 
you're mind let me know and I'll bet you a dev board.

Thanks again and have a great weekend!

Jay

JChristensen, 2018-02-03:
Jay, a little while ago I pushed version 1.1.1 with fixes that allow the library to compile for ARM architecture. Whether it will run properly, I have no idea. I would be interested in any reports one way or the other if you do try it. If there are issues with ARM platforms, I may have a look but no promises at this time. I am still advertising the library for AVR only so the IDE will give a compile warning. When using the library with an ARM board, it is necessary to instantiate a DS3232RTC object, e.g. `DS3232RTC myRTC;`

1cc386ea8e470a, 2018-02-03:
I'll give it a whirl today and let u know how it goes!

Thanks a million!
Jay