JChristensen/DS3232RTC #25: DS3232RTC and SD libraries clashing

nandakrpp, 2016-03-29:
Hello Jack,

First of all, thanks for DS3232RTC library. It works great on my system that wakes up Arduino Mega 2562 from sleep. I am using the alarm interrupt code mentioned in https://github.com/JChristensen/DS3232RTC/issues/5 without a problem on its own. But the alarm is not triggered after I implemented in SD card library. I was wondering if you have encountered the same problem.

Regards.

JChristensen, 2016-03-30:
@nandakrpp, I am not aware of such an issue. It's a bit surprising, with the RTC using the I2C bus and the SD card using the SPI bus, I wouldn't think there would be a lot of opportunity for clashing. Be sure that interrupts are not disabled, else the Wire library will not work.
If there is a demonstrable issue with the library, please submit a [SSCCE](http://www.sscce.org/).

closed 2016-03-30 by JChristensen

nandakrpp, 2016-04-05:
@JChristensen, you are absolutely right. The problem is not between RTC's I2C and SD's SPI buses. I found out that the Ardafruit BMP180 (used with BMP085 library), another I2C device on the same bus has caused the problem. A strange thing is that there is also another I2C device on the same bus – TSL2591 that works just fine with the RTC library. Both sensors are initialised and called to read data one after another after the Wire library is started.

`void setup()
{
// initialise Wire library
// initialise BMP library
// initialise TSL2591 library
// initialise RTC and set time for alarm 1 
}`

Without initialising BMP, everything works fine together. Any pointer to solve this problem is much appreciated.

Thanks.