JChristensen/DS3232RTC #26: Getting the Alarm time and type

CalDexter, 2016-04-02:
This is really an amazing library and it helps me a lot to make my first steps with an Arduino UNO and a DS3131 board. However, I have missed the feature to receive the alarm time once set. I would suggest to include a further function to your library doing this.

Here is a hack of myself doing the required task:
```C
/*----------------------------------------------------------------------*
 * Reads the current alarm time for alarm from the RTC and returns it   *
 * in a tmElements_t * structure tm and an ALARM_TYPES_t variable       *
 * alarmType. I am afraid that tmElements_t is somehow abused...        *
 *----------------------------------------------------------------------*/
byte DS3232RTC::readAlarm(int alarm, tmElements_t &tm, ALARM_TYPES_t &alarmType)
{
  byte s=0, m,h,d, aType;
  if (alarm == ALARM_1) {
    aType = 0x00;
    i2cBeginTransmission(RTC_ADDR);
    i2cWrite((uint8_t)ALM1_SECONDS);
    if ( byte e = i2cEndTransmission() ) return e;    
    //request 4 bytes (secs, min, hr, dow)
    i2cRequestFrom(RTC_ADDR, 4);
    s = i2cRead();
  } else {
    aType = 0x80;
    i2cBeginTransmission(RTC_ADDR);
    i2cWrite((uint8_t)ALM2_MINUTES);      
    if ( byte e = i2cEndTransmission() ) return e;    
    i2cRequestFrom(RTC_ADDR, 3);
  }
  m = i2cRead();
  h = i2cRead();
  d = i2cRead();
  // Decode alarm type
  aType |= (s & _BV(7)) ? 0x01 : 0;    
  aType |= (m & _BV(7)) ? 0x02 : 0;
  aType |= (h & _BV(7)) ? 0x04 : 0;
  aType |= (d & _BV(7)) ? 0x08 : 0;
  // set time / day / date values
  tm.Second = bcd2dec(s & 0x7f);
  tm.Minute = bcd2dec(m & 0x7f);
  tm.Hour = bcd2dec(h & 0x7f);    //  assumes 24hr clock
  tm.Wday = (d & _BV(DYDT)) ? 0 : bcd2dec(d & 0x3f);
  tm.Day = (d & _BV(DYDT)) ? bcd2dec(d & 0x3f) : 0;
  tm.Month = 0;
  tm.Year = 0;
  alarmType = (ALARM_TYPES_t)aType;
  return 0;
} 
```

I am quite new for programming Arduino code and thus this code is probably not bullet proof, not well-suited to include, and possibly not well enough tested. I am sorry for that, however, it has worked for myself. Possibly, there is a strong reason not to include something like this to the code.

Best regards and many thanks again for this nice library,

Andreas

JChristensen, 2016-04-04:
Hello Andreas, thanks for your suggestion. As you saw there was a previous suggestion, so I will consider adding this functionality due to popular demand ;-)  I'm mostly concerned with code footprint but if a function is not called then it should not affect the size of the image in program memory.
Unfortunately I can't address this very soon, so best to go ahead and do your own thing for now.
Cheers ... Jack

quiquelhappy, 2021-06-17:
Please, add this to the library