JChristensen/DS3232RTC #29: ESP8266 compatibility issues??

stephan14x, 2016-05-24:
Hi,

this is my first try with the library and a NodeMCU Board (ESP12E) using Arduino IDE 1.6.8

As soon as I'm including the Library and flash even the simplest sketch:

`#include <DS3232RTC.h>    //http://github.com/JChristensen/DS3232RTC
#include <TimeLib.h>      //https://github.com/PaulStoffregen/Time 
#include <Wire.h>         //http://arduino.cc/en/Reference/Wire 

int loops = 0;

void setup() {
  Serial.begin(115200);
  delay(10);
}

void loop()
{
  Serial.print("Loop: ");
  Serial.println(loops);

  loops++;
}`

the ESP fails to run and I'm getting this on the serial console:

`ets

 Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld`

Tried this with the RTC Module connected and without it and I'm not really sure what pins to use either. Also tried the example sketches without luck.

Does anyone have any experience with the ESP8266 and a DS3231.

Stephan

JChristensen, 2016-05-24:
Please raise an issue here only if there is a demonstrable and repeatable issue with the library and be sure to include a [SSCCE](http://www.sscce.org/). I don't have an ESP8266 so cannot help. For help debugging your project, please use the Arduino forum, etc. I'm not up on Paul Stoffregen's Time library; unless it's backward compatible with the old Arduino Time library, there may be some issues. Paul has published some improved libraries that aren't always 100% backward compatible. Not that that's a bad thing; he certainly does fine work, just saying. Good luck with your project.

closed 2016-05-24 by JChristensen

bHogan2, 2016-12-23:
FWI, I can confirm that your lib doesn't work on the ESP with the 2.3.0 ESP core. (earlier cores worked).
Just by including it it causes a WDT error. Often this happens on the ESP when these is a while loop or the like that goes on too long. A yield() or delay(0) in the loop generally fixes these problems by allowing the housekeeping to take place.

Given that you don't use the ESP I can understand your position, but since it is so popular you may want to note this somewhere to save some grief.

(The DS3231 lib on Adafruit does work.)