JChristensen/DS3232RTC #33: Using SqWav 1Hz to interrupt Arduino every second

codemonkey09, 2016-09-20:
I am attempting to build an RTC using the DS3231. In my code i use the 1Hz square wave to interrupt the Arduino every second, and read the time from the 3231 over i2c every time the interrupt is received. The problem is that, I have noticed that my clock is drifting over time. So far, over 3 days (72hrs) the clock has sped up by around 2.5 seconds. I am unable to identify the source of the issue, but, based on your other comments i am guessing the 1Hz interrupt is not a general use-case. 
I feel like i am missing something here. Any help would be appreciated.

JChristensen, 2016-09-20:
My first question is what is the source of this particular DS3231. Devices from eBay or other sketchy sellers may well be counterfeit and out of spec (or worse). I recommend only buying from reputable suppliers that can trace the supply chain back to the manufacturer.
In fact I often use the 1Hz interrupt for timekeeping and it works just fine. Although I do it a little differently. I only read the time from the RTC once during initialization, then I increment it at each interrupt. Doing an I2C read every second is just unnecessary overhead.

closed 2016-09-20 by JChristensen

codemonkey09, 2016-09-21:
yes, this part was bought off eBay. I can try another part from DigiKey. Thanks for the tip.