JChristensen/DS3232RTC #37: Please initiate the i2c communication during Arduino Setup to avoid wdt resets on ESP8266

gargamel007, 2017-01-16:
Hi, 
When using this lib with ESP8266 Arduino I'm getting watchdog resets even before the setup(); function. I've narrowed it to the call the RTC constructor that gets called on loading the RTC library.
```cpp
DS3232RTC::DS3232RTC()
{
    i2cBegin();
}
```
I did manage to solve this by creating another function to call in my setup() function and begin the Wire protocol.
```cpp
DS3232RTC::DS3232RTC()
{
  //Does nothing !
}

/*----------------------------------------------------------------------*
 * Call in the setup() function to start protocol                       *
 *----------------------------------------------------------------------*/
void DS3232RTC::begin()
{
    i2cBegin();
}
```
I don't know if this is the best fix as I'm just beginning with ESP8266 chips, but this might help others !

JChristensen, 2017-01-17:
I don't have an ESP8266, so I have no way to re-create this issue or debug it. Sorry.

GanjaFarmer, 2018-02-26:
I support this. You made a recent change so "RTC" isn't created for "non AVR" platform. I don't know if it's related to this issue and I really think I2C should be initialized inside setup(), no matter the platform. How do we (ESP866/32 users) create a new DS3232RTC object, without it calling Wire.begin? Please implement the change that ioExpander explained :)

JChristensen, 2018-02-28:
I agree that it's better to put the I2C initialization in a begin() function, but the proposed solution would break backward compatibility which I'd rather not do. I'm working on a slightly different approach but it's giving me some trouble.

JChristensen, 2018-03-03:
If someone could test the [dev branch](https://github.com/JChristensen/DS3232RTC/tree/dev) on an ESP8266, that'd be great. For architectures other than AVR, the user's code will need to instantiate a DS3232RTC object. The constructor now has a bool parameter to indicate whether to initialize the I2C bus in the constructor, this parameter defaults to true. A begin() member function has been added to alternately initialize the I2C bus when & where the user desires.

To summarize: (1) Instantiate a DS3232RTC object, e.g. `DS3232RTC myRTC(false;)` and then, in setup or wherever, `myRTC.begin();`

GanjaFarmer, 2018-03-03:
I like it :) It's obviously working. I prefer to just call Wire.begin instead of rtc.begin, because I use other I2C devices. Thanks!

JChristensen, 2018-03-03:
That makes sense with multiple devices on the bus. Thanks for the test!

commit 5be912a9 (2018-03-03): Instantiate RTC object for AVR architecture only.
Add parameter to constructor to indicate whether to
initialize the I2C bus (default is true). Closes #37.
closed 2018-03-03 by JChristensen