JChristensen/DS3232RTC #43: Bootloading Error with ESP8266

davescherler, 2017-08-21:
Hi There, 

I've been working on adding a DS3231 RTC to a project that's running Arduino on an ESP8266 (I will add Wifi later). I've had success with the DS1307 but I wanted to make use of the alarm functions that the DS3231 provide. 

I've been able to set and read the time using Adafruit's RTClib but I've moved on to your library  since the documentation is so good and a lot of folks seem to have luck with setting and handling the alarms. My apologies if this isn't the appropriate place to post this kind of issue but I've already scoured the web and posted this issue on the Arduino forums but I haven't had much luck yet! 

My Hardware: 
ESP8266 running Arduino 1.6.12
DS3231 SDA > pin 4 on ESP
DS3231 SCL > pin 5 on ESP
DS3231 SQW > 10K pullup resistor to VCC & pin 14 on ESP

My Issue: the sketch below complies and uploads but immediately throws the following errors in the Monitor: 

"ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset" 

Then nothing. I can always go back and load different sketches and they all compile and run successfully so I don't think there's a wiring issue. I've also deleted and redownloaded the DS3232RTC library to make sure my copy wasn't corrupted. I've been googling those error messages and a lot of the solutions I'm finding relate to wiring and insufficient power issues. But if I'm able to load different sketches with the same wiring/powering setup, that shouldn't be an issue, correct? I'm out of ideas. 

Has anyone run into this issue before? Is it related to my other libraries? It must be related to my sketch but I can't figure out where exactly. Any help is greatly appreciated!! 

```
#include <Time.h>
#include <TimeLib.h>
#include <Wire.h>
#include <DS3232RTC.h>

volatile boolean alarmFlag = false; 

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  //delay(1000);
  Serial.println("initializing DS3231");
  setSyncProvider(RTC.get);
  RTC.squareWave(SQWAVE_NONE);
  RTC.alarmInterrupt(ALARM_1, false);
  RTC.alarmInterrupt(ALARM_2, false);
  RTC.setAlarm(ALM1_MATCH_SECONDS, 20, 0, 0, 2);
  RTC.alarmInterrupt(ALARM_1, true);
  pinMode(14, INPUT);
  attachInterrupt(14, alarmISR, FALLING); 
}

void loop() {
  time_t currentTime;
  currentTime = RTC.get();
  Serial.println(currentTime);
  if (alarmFlag == true) {
    Serial.println("the 20 second alarm just went off!");
    alarmFlag = false;
  }
//delay(1000);
}

void alarmISR() {
  alarmFlag = true;
}
```

JChristensen, 2017-08-21:
Sorry, I can't help, I don't have an ESP8266 so no way to duplicate the problem. I don't know where the "wdt reset" message is coming from, it's not in your code and not in the library.

closed 2017-08-21 by JChristensen

davescherler, 2017-08-22:
Hi @JChristensen thanks for the response. I understand. I was eventually able to achieve what I wanted with alarms using a different library. Though my code not loading using this library still bothers me so when I have some time I'll try to debug further. Since you mention those error messages aren't defined in your library, that at least tells me some lower level MCU thing is failing to run at boot time, so that's helpful. If I find a solution and think it's worth sharing, I'll do so here. Thanks!

davescherler, 2017-08-23:
Hi @JChristensen Just wanted to pass along an update since this was really bizarre. I was really beating my head against the wall trying to figure this loading issue so I decided to (tediously) add in my code to a new sketch line by line. For whatever reason, the sketch now loads. Literally not a single line of code or wiring changed! The only thing I can remember doing is - in an act of desperation - I renamed the local Time.cpp file to TimeLib.cpp and back again. I guess it's possible that that change could have had some affect but I'm just glad it's working! 

Could I bother you for one **really** quick question? The documentation for setting alarms states that the `RTC.setAlarm` method requires a day of the week parameter between 1-7 (or something like "dowTuesday"). But what if I want a 20 second alarm to go off every day? Or just more than that one day that I pass in? Will the method accept multiple values there akin to something like: 

`RTC.setAlarm(ALM1_MATCH_SECONDS, 20, 0, 0, dowMonday | dowTuesday | dowWednesday | dowThursday | dowFriday); //weekdays only`

JChristensen, 2017-08-23:
The really quick answer is no. Days of the week cannot be ORed together like that. I'm also not sure what you mean by "a 20 second alarm". If that is an alarm that goes off for 20 seconds and then resets, the RTC does not work that way.

There is a tendency to think of the alarms like those in an alarm clock that we might have on our nightstand. They're really not a lot like that. Read the Alarms section in the datasheet closely. The alarm is raised as long as the time registers match the alarm registers, taking the mask bits into account. Hence, there is no such thing as an alarm that lasts for 20 seconds, and no such thing as a weekday only alarm.

Of course such things are possible, but additional application code will be required. Typically I will set the alarm register, and when the alarm is raised, the application will calculate the next desired alarm time and set the alarm register accordingly.

The way the alarm registers work, there are only a few special cases where they can be set once and forgotten (again, see the datasheet). If I were using these RTCs for a common alarm clock, the alarm times presented by the user interface would not be the same as the contents of the alarm register! In fact there might be little point in even using the alarm registers, it might be just as good to maintain the alarm information in the application only. HTH.