JChristensen/DS3232RTC #44: ALM1_MATCH_HOURS

leobel96, 2017-09-12:
Hello. I'm sorry but I don't understand how to set an alarm1 that matches hours, minutes and seconds using the type ALM1_MATCH_HOURS. Which is the fields' order inside the function? Thank you

JChristensen, 2017-09-12:
Order of the function arguments is given [here](https://github.com/JChristensen/DS3232RTC#setalarmalarm_types_t-alarmtype-byte-seconds-byte-minutes-byte-hours-byte-daydate). When using ALM1_MATCH_HOURS, the dayOrDate parameter can be zero.

closed 2017-09-12 by JChristensen

leobel96, 2017-09-12:
Perfect. Thank you!

leobel96, 2017-09-13:
Sorry but I have another question: after the alarm the interrupt pin falls; then it goes high again automatically or I have to do it with a function? Thank you.

JChristensen, 2017-09-13:
The short answer is that the alarm flag is set as long as the time registers match the alarm register, allowing for the mask bits. Please read the Alarms section in the datasheet. People tend to think this RTC functions like the typical alarm clock that we may have on our nightstand, and it's really not so much like that. Best to read and fully comprehend the datasheet, and suppress any preconceived notions of how an "alarm" should act.

leobel96, 2017-09-13:
Ok, I'll read It!