JChristensen/DS3232RTC #45: How about an example to create multiple alarms and triggers? thank you

husainaw, 2017-09-16:
How about an example to create multiple alarms and triggers? sorry i really have trouble understanding thank you

JChristensen, 2017-09-17:
I'm considering writing a short primer on alarms because they seem to be poorly understood. I'd like to understand more about what sort of examples you have in mind. What do you mean exactly by "multiple alarms", and what is a "trigger"? Is a trigger different from an alarm? These RTCs have two alarms so it seems that "multiple" can only mean one or two.

husainaw, 2017-09-17:
I am very happy and feel very helpful if you are willing to provide further documentation sir.
what I mean by multiple alarms is for example I want to create 40 different alarm types and turn on the lights on schedule of 40 alarms repeated every day at the same time and hour.
I've tried to use library timeAlarm, but on 4th alarm and the nex,t does not work properly.

#include <DS3232RTC.h> //http://github.com/JChristensen/DS3232RTC
#include <TimeLib.h>
#include <Wire.h> // https://github.com/PaulStoffregen/TimeAlarms
#include <TimeAlarms.h> //https://github.com/PaulStoffregen/TimeAlarms
int lamp = 5;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  pinMode(lamp, OUTPUT);
  digitalWrite(lamp, HIGH); // default active low

  setSyncProvider(RTC.get);   // the function to get the time from the RTC
  if (timeStatus() != timeSet)
    Serial.println("Unable to sync with the RTC");
  else
    Serial.println("RTC has set the system time");
  //9AM
  Alarm.alarmRepeat(9, 0, 0, lampON);
  Alarm.alarmRepeat(9, 0, 46, lampOFF);
  Alarm.alarmRepeat(9, 15, 0, lampON);
  Alarm.alarmRepeat(9, 15, 46, lampOFF);
  Alarm.alarmRepeat(9, 32, 0, lampON);
  Alarm.alarmRepeat(9, 32, 46, lampOFF);
  Alarm.alarmRepeat(9, 34, 0, lampON);
  Alarm.alarmRepeat(9, 34, 46, lampOFF);
  Alarm.alarmRepeat(9, 45, 0, lampON);
  Alarm.alarmRepeat(9, 45, 46, lampOFF);
  //10AM
  Alarm.alarmRepeat(22, 3, 0, lampON);
  Alarm.alarmRepeat(22, 3, 46, lampOFF);
  Alarm.alarmRepeat(10, 10, 0, lampON);
  Alarm.alarmRepeat(10, 10, 46, lampOFF);
  Alarm.alarmRepeat(10, 20, 0, lampON);
  Alarm.alarmRepeat(10, 20, 46, lampOFF);
  Alarm.alarmRepeat(10, 30, 0, lampON);
  Alarm.alarmRepeat(10, 30, 46, lampOFF);
  Alarm.alarmRepeat(10, 40, 0, lampON);
  Alarm.alarmRepeat(10, 40, 46, lampOFF);
  Alarm.alarmRepeat(10, 50, 0, lampON);
  Alarm.alarmRepeat(10, 50, 46, lampOFF);
  //11AM
  Alarm.alarmRepeat(11, 00, 0, lampON);
  Alarm.alarmRepeat(11, 00, 46, lampOFF);
  Alarm.alarmRepeat(11, 10, 0, lampON);
  Alarm.alarmRepeat(11, 10, 46, lampOFF);
  Alarm.alarmRepeat(11, 20, 0, lampON);
  Alarm.alarmRepeat(11, 20, 46, lampOFF);
  Alarm.alarmRepeat(11, 30, 0, lampON);
  Alarm.alarmRepeat(11, 30, 46, lampOFF);
  Alarm.alarmRepeat(11, 40, 0, lampON);
  Alarm.alarmRepeat(11, 40, 46, lampOFF);
  Alarm.alarmRepeat(11, 50, 00, lampON);
  Alarm.alarmRepeat(11, 50, 46, lampOFF);
  //12AM
  Alarm.alarmRepeat(12, 0, 0, lampON);
  Alarm.alarmRepeat(12, 0, 46, lampOFF);
  Alarm.alarmRepeat(12, 10, 0, lampON);
  Alarm.alarmRepeat(12, 10, 46, lampOFF);
  Alarm.alarmRepeat(12, 20, 0, lampON);
  Alarm.alarmRepeat(12, 20, 46, lampOFF);
  Alarm.alarmRepeat(12, 30, 0, lampON);
  Alarm.alarmRepeat(12, 30, 46, lampOFF);
  Alarm.alarmRepeat(12, 40, 0, lampON);
  Alarm.alarmRepeat(12, 40, 46, lampOFF);
  Alarm.alarmRepeat(12, 50, 0, lampON);
  Alarm.alarmRepeat(12, 50, 46, lampOFF);
  //13AM
  Alarm.alarmRepeat(13, 0, 0, lampON);
  Alarm.alarmRepeat(13, 0, 46, lampOFF);
  Alarm.alarmRepeat(13, 10, 0, lampON);
  Alarm.alarmRepeat(13, 10, 46, lampOFF);
  Alarm.alarmRepeat(13, 20, 0, lampON);
  Alarm.alarmRepeat(13, 20, 46, lampOFF);
  Alarm.alarmRepeat(13, 30, 0, lampON);
  Alarm.alarmRepeat(13, 30, 46, lampOFF);
  Alarm.alarmRepeat(13, 40, 00, lampON);
  Alarm.alarmRepeat(13, 40, 46, lampOFF);
  Alarm.alarmRepeat(13, 50, 0, lampON);
  Alarm.alarmRepeat(13, 50, 46, lampOFF);
  //14AM
  Alarm.alarmRepeat(14, 0, 0, lampON);
  Alarm.alarmRepeat(14, 0, 46, lampOFF);
  Alarm.alarmRepeat(14, 10, 0, lampON);
  Alarm.alarmRepeat(14, 10, 46, lampOFF);
  Alarm.alarmRepeat(14, 20, 0, lampON);
  Alarm.alarmRepeat(14, 20, 46, lampOFF);
  Alarm.alarmRepeat(14, 30, 0, lampON);
  Alarm.alarmRepeat(14, 30, 46, lampOFF);
  Alarm.alarmRepeat(14, 40, 0, lampON);
  Alarm.alarmRepeat(14, 40, 46, lampOFF);
  Alarm.alarmRepeat(14, 50, 0, lampON);
  Alarm.alarmRepeat(14, 50, 46, lampOFF);
  //15AM
  Alarm.alarmRepeat(15, 0, 0, lampON);
  Alarm.alarmRepeat(15, 0, 46, lampOFF);
  Alarm.alarmRepeat(15, 10, 0, lampON);
  Alarm.alarmRepeat(15, 10, 46, lampOFF);
  Alarm.alarmRepeat(15, 20, 0, lampON);
  Alarm.alarmRepeat(15, 20, 46, lampOFF);
  Alarm.alarmRepeat(15, 30, 0, lampON);
  Alarm.alarmRepeat(15, 30, 46, lampOFF);

}

void loop() {
  // put your main code here, to run repeatedly:
  digitalClockDisplay();
  Alarm.delay(1000);
}

void lampON() {
  digitalWrite(lamp, LOW);
  Serial.println("lamp ON");
}

void lampOFF() {
  digitalWrite(lamp, HIGH);
  Serial.println("lamp OFF");
}

void digitalClockDisplay()
{
  // digital clock display of the time
  Serial.print(hour());
  printDigits(minute());
  printDigits(second());
  Serial.println();
}

void printDigits(int digits)
{
  // utility function for digital clock display: prints preceding colon and leading 0
  Serial.print(':');
  if (digits < 10)
    Serial.print('0');
  Serial.print(digits);
}

JChristensen, 2017-09-17:
Sounds like this is more of an issue with your application, possibly with how the TimeAlarms library is being used. I haven't used that library, so I can't comment. I will say that in this sort of application I'm not sure I'd use the RTC alarms at all. It doesn't make sense to me to try and shoehorn 40 different events into the RTC alarms.

I'll close this issue for now but watch this repo for a primer on RTC alarms with some example sketches.

PS: I had a very quick look at Paul Stoffregen's TimeAlarms library and it looks like that by default only 6 alarms are allowed on an AVR platform, so 40 may be over-achieving ;-)

closed 2017-09-17 by JChristensen

husainaw, 2017-09-17:
thank you for your attention and explanation, I will learn more again. :)