JChristensen/DS3232RTC #48: Problem with setting the alarm on a given day of the week

piolin798, 2017-10-24:
Hi, I´m trying to set an alarm so it is triggered at a specific day of the week e.g. every monday, using function "setAlarm1". However, when I open the serial console, day appears as "unknown". 

  // Set Alarm - 10h:45m:30s in every Friday (1 - Mon, 7 - Sun)
  //setAlarm1(Date or Day, Hour, Minute, Second, Mode, Armed = true)
  clock.setAlarm1(7, 15, 40, 15, DS3231_MATCH_DY_H_M_S);

Please tell me how this could be solved.

JChristensen, 2017-10-24:
That call to setAlarm() is totally incorrect for this library. It should be RTC.setAlarm() not clock.setAlarm(). Also, the arguments are incorrect. See https://github.com/JChristensen/DS3232RTC#alarm-methods

closed 2017-10-24 by JChristensen

JChristensen, 2017-10-24:
Further, there is no function "setAlarm1", there is only "setAlarm". I think what you want is:
```c++
// set alarm 1 for 10:45:30 every Friday
RTC.setAlarm(ALM1_MATCH_DAY, 30, 45, 10 dowFriday);
```