JChristensen/DS3232RTC #49: 'RTC' was not declared in this scope

Robiv8, 2018-03-30:
sketch_mar30a:13: error: 'RTC' was not declared in this scope
     setSyncProvider(RTC.get);   // the function to get the time from the RTC
                     ^
exit status 1
'RTC' was not declared in this scope

JChristensen, 2018-03-30:
What board are you using?

JChristensen, 2018-03-30:
Do you have the latest release of the library (1.2.0)?

Robiv8, 2018-03-30:
Hallo,
Yes, I downloaded it here a few hours ago.
Board is NodeMCU 1.0 (ESP-12E) Module its a ESP8266MOD Amica

JChristensen, 2018-03-30:
For architectures other than AVR, you need to instantiate an RTC object, for example:
```c++
DS3232RTC myRTC;
```
Then use that object for all library calls, for example:
```c++
setSyncProvider(myRTC.get);
```

closed 2018-03-30 by JChristensen

Robiv8, 2018-03-30:
Thank you very much for the Info

JChristensen, 2018-03-31:
The documentation was lacking regarding non-AVR architectures and has now been updated with release 1.2.1.

rtek1000, 2019-10-04:
Need update the example TimeRTC:

> // Arduino DS3232RTC Library
> // https://github.com/JChristensen/DS3232RTC
> //
> // Example sketch illustrating Time library with Real Time Clock.
> // This example is identical to the example provided with the Time Library,
> // only the #include statement has been changed to include the DS3232RTC library.
> 
> 
> #include <DS3232RTC.h>      // https://github.com/JChristensen/DS3232RTC
> 
> DS3232RTC RTC;  // <--------------------- "Added"
> 
> void setup()
> {
>     Serial.begin(9600);
>     setSyncProvider(RTC.get);   // the function to get the time from the RTC
>     if(timeStatus() != timeSet)
>         Serial.println("Unable to sync with the RTC");
>     else
>         Serial.println("RTC has set the system time");
> }
> 
> void loop()
> {
>     digitalClockDisplay();
>     delay(1000);
> }
> 
> void digitalClockDisplay()
> {
>     // digital clock display of the time
>     Serial.print(hour());
>     printDigits(minute());
>     printDigits(second());
>     Serial.print(' ');
>     Serial.print(day());
>     Serial.print(' ');
>     Serial.print(month());
>     Serial.print(' ');
>     Serial.print(year());
>     Serial.println();
> }
> 
> void printDigits(int digits)
> {
>     // utility function for digital clock display: prints preceding colon and leading 0
>     Serial.print(':');
>     if(digits < 10)
>         Serial.print('0');
>     Serial.print(digits);
> }

rtek1000, 2019-10-04:
https://github.com/JChristensen/DS3232RTC/blob/master/examples/TimeRTC/TimeRTC.ino

JChristensen, 2019-10-04:
@rtek1000 why do you reply to a closed issue? What board are you using?
The need to define a DS3232RTC object for non-AVR architectures is documented [here in the README file](https://github.com/JChristensen/DS3232RTC#constructor).

kp1851, 2023-10-10:
Thanks. I add "DS3232RTC RTC;" and it worked.