JChristensen/DS3232RTC #50: Make API consistent across architectures

bperrybap, 2018-04-06:
Here are some suggestions that could make things consistent across all architectures.
I've been wrestling with this recently as I'm converting a project that used a Teensy2 to use an ESP8266 WeMos D1 mini.

Note: if you like the suggestions below, I can create a pull request.

The API is non consistent across Arduino core architectures.
This is something that really should be avoided.
This is a result of several recent commits and some of the usage documentation.
Btw, I think some of the documentation is likely to be confusing for people
For example I'm guessing that many will find this confusing: 

> the user should not use the constructor for AVR boards. For other architectures, the user's code must instantiate a DS3232RTC object.

I get how it works, on AVR you can use the RTC object but on non AVR you must declare one, but  the average user will likely not grasp this
Also, to keep the "rtc" object names the same within a sketch that must run on multiple architectures, it requires having a conditional to declare an object named RTC only on none AVR platforms.
Or to always create an "rtc" object and just abandon the predefined one named RTC on AVR.

I think that there are better ways to handle the backward compatibility issues as well as the need to move Wire.begin() away from the constructor.
As a FYI, in case you didn't know, if you call Wire.begin() from a constructor on the esp8266 platform, it will crash the board and since it is so early during initalization it is difficult to upload another sketch to it as it the board is rapidly crashing and tearing down the USB connection.
For a while I thought the board was bricked. I was able to correct it, but only on a certain PC host and even then it took many attempts to get an upload to work.

Here is an alternative that I think works better and would make things simpler and consistent across platforms:

- Always create the default RTC object
(Its being created on AVR so go ahead and create it on all platforms)
- Change the default initI2C parameter to default to false
- Create the RTC object differently for AVR vs other platforms.
- keep the existing begin() code as is.

i.e.

in the header file:
```
        DS3232RTC(bool initI2C = false);
```

The reason being that new code should never initialize the i2c h/w in the constructor so why make all the new code that declares an object specify that initialization should not be done?

in the .cpp file:
```
#if defined(__AVR__)
DS3232RTC RTC(true);      //instantiate an RTC object & initialize Wire for backward compatibility
#else
DS3232RTC RTC(false);      //instantiate an RTC object
#endif
```
note: using __AVR__ is potentially better than using ARDUINO_ARCH_AVR as ARDUINO_ARCH_AVR is not available in older IDEs.

This should provide the best of all worlds.
- all existing AVR code continues to work
- Non AVR code can also use a pre defined RTC object that does not initialize Wire until you call RTC.begin()
- All new code should call RTC.begin() including new AVR code
(it does not hurt to call Wire.begin() more than once)
- While not necessary, new code could declare their own "rtc" object with any name other than RTC just like today.
(The existing RTC object would still exist but not be used)
- by making the default value of initI2C false, sketches that declare an "rtc" object don't have to pass in a parameter - makes declaration simpler.
- esp8266 sketches can still override the i2c pins used if they simply call Wire.begin(sda_pin, scl_pin) before calling RTC.begin()

I think this would make things simpler and consistent across platforms.
I don't think it will affect any existing code, including code that is using the current API.
(Other than sketches that created an "rtc" object name "RTC" on non avr platforms)
It can make the documentation simpler since the documentation would now be consistent across all platforms.
All sketches on any h/w platform can use the built in predefined RTC object, or can define their own with another name.
All sketches should call the begin() function including on AVR platforms.

What this means for examples, is that all examples should now call RTC.begin() first.
But that will allow all the examples to work on non avr platforms as well.
For best compatibility with ESP8266 platforms pin 3 would be better than pin 2  as several of them use GPIO2 for the LED or i2c. none use GPIO3 for anything so it would be a better choice.

JChristensen, 2018-04-07:
Whew. Well those are all good ideas. My main objective was to stay backwards compatible for AVR while providing functionality for other architectures even if the API wasn't identical. While that may be somewhat annoying for folks working with multiple architectures, practically I'm not sure how big of an issue it is. But still I get the point.

There is one other issue that I'm aware of with the SAM architecture chips (Arduino Due, Zero) and that is a name collision with "RTC". So the following would still cause an error for those boards:
```c++
#if defined(__AVR__)
DS3232RTC RTC(true);      //instantiate an RTC object & initialize Wire for backward compatibility
#else
DS3232RTC RTC(false);      //instantiate an RTC object
#endif
```

For a third-party library like this, I'm not even sure I like the idea of instantiating an RTC object. I'm having second thoughts on backwards compatibility and wondering if a consistent API isn't more important. The user would be responsible for defining the RTC object and initializing I2C on all architectures. What are your thoughts on that?

bperrybap, 2018-04-07:
I tend to agree that a consistent API is very important.
Me personally, I don't have a huge preference for whether or not there is a predefined object.
I just want to have a way to compile sketches for multiple cores without having to use ifdefs in the sketch.

Moving to full consistency and maintaining full backward compatibility isn't really possible given the need to move Wire.begin() out of the constructor and the existing name collision issue with the SAM core.
That said, my suggestion would be to have a consistent API by either always having a predefined "RTC" object or never having one.
I took a look at the issue on the SAM core and it is still possible to make a predefined object name RTC work on that core. It is a bit ugly.
Since RTC is a define in the core, you could undefine it in the DS3232RTC header file.
i.e. something like this:
```
#undef RTC
extern DS3232RTC RTC;
```
While that #undef addition to DS3232RTC.h along with the RTC object declaration above added to the .cpp file does make the API consistent, and provide 100% full backward compatibility with existing AVR code, it doesn't make the API backward compatible on non AVR cores to what worked on AVR cores. i.e. new code must always call RTC.begin() to have a consistent API.

That said, the issue of name collision with the name "RTC" could show up again. So while it works now, it could break again in the future and there might not be a work around.

Since it requires a sketch code changes (adding calling RTC.begin() ) to make the sketch API compatible across all cores, a user could also easily add a rtc object declaration at the same time should there be no pre defined RTC object. Not having a pre defined RTC object definitely would futureproof the library against name collisions but would break backward compatibility.

Here are some alternatives:

1) do the #undef hack & pre-init RTC on AVR platforms.
This provides a  predefined RTC object along with a consistent API (users must call RTC.begin())and also provides full backward compatibility with existing AVR code. It is minimal impact on examples and existing documentation.
It does have a risk that if it breaks in the future, there may be no work around to fix it.

2) Change default constructor to false and  predefine RTC object only for AVR
(basically what is there already other than changing the default constructor parameter)
i.e. predefine a RTC object on AVR that does Wire library initialization.
The predefined RTC object is only there for existing AVR code backward compatibility and would be undocumented.
This can  provide a consistent API, and backward compatibility with existing AVR code.
The backward compatibility for existing AVR code would not be documented.
The documentation and examples would be need to be updated showing declaring an "rtc" object and calling rtc.begin() first.
Also, documentation would need to indicate that the name "RTC" cannot be used as an rtc object name.

This works  because it creates a pre-defined RTC object on AVR platforms but it is just there for the undocumented backwards compatibility. 
All new code would declare their own object (named something other than RTC) and the predefined RTC object on AVR will get removed by the linker as there would be no references to it.
The reason to change the default constructor to false, is so that new code can declare their rtc object as:
DS3232RTC myrtc;

3) Remove pre declaration of an RTC object completely and require rtc.begin()
This provides a clean consistent API with no risk of name collisions in the future.
It does break 100% of any existing AVR code using the library that depends on a predefined RTC object.
It requires changing the documentation and examples to reflect that an rtc object must be declared and that rtc.begin() must be called first.

-------------------------------------------------------

Even though option 1 can offer a predefined RTC object along with a consistent API, it does have the risk that if there is an issue in the future with RTC name collision, it may not be solvable and require changing the API.

Option is 2 is kind of nice but is a bit funky in that users cannot use the name "RTC" for their own object on AVR platforms due to the hidden RTC object for backward compatibility and also cannot use it on SAM platforms due to the name collision, but that is a documentation issue.

Option 3 is the same as option 2 but without the pre defined RTC object for AVR.
It is by far the cleanest looking forward but does break all existing AVR code.

If options 2 or 3 are chosen, then it would probably be best to move the library version to 2.x to indicate that there is an API change.

One possibility could be to do option 1 as the last 1.x release of the library.
It provides a consistent API with a predefined RTC object for all cores.
Not quite the same as the existing AVR api is it requires using RTC.begin() on non AVR cores but it does provide a way to a sketch that works on all cores with no ifdefs if RTC.begin() is always used first including on AVR cores.
Then come out with a 2.0 release where there is no pre defined RTC object anymore.
i.e. try to provide an API with a predefined RTC object on all cores as a last 1.x release
and then move away from having a predefined RTC object with a 2.0 release.
But it is more work than just jumping straight to option 3 with a 2.0 release.

---------------------------

[... cut ...]
The advantage of switching to this would be that library could  be a hybrid of all the options and the code is built based on a define in the sketch.
i.e. by default the code does not provide backward compatibility with existing old AVR code but the sketch could turn a define to re-enable it.

---------------------------------
No real easy answer. But if you considering moving to no longer pre defining an RTC object to simplify things, then option 3 and bumping to version 2.x would be the way go. Users that still need a library for their existing old AVR code, could always install a 1.x version of the library.

JChristensen, 2018-04-07:
I was also thinking of a 2.0.0 release, basically option 3. In that case, I probably wouldn't bother with another 1.x release first unless a lot of complaints come in with 1.2.2. The library manager is a great help, folks can continue to use 1.2.2 until and unless they are ready to move to 2.0.0. And surely the effort required to upgrade application code for 2.0.0 would be minimal anyway.

Next in my queue is to revamp my MCP79412RTC library. I've already started (just) working on that. So maybe I will go with the same approach there (a 2.0.0 release) and see how that is received (although it's not as popular a library). That'll give some time for DS3232RTC 1.2.2 to settle a little more to see if anything else is needed before moving it to 2.0.0.

Appreciate your input!

bperrybap, 2018-04-07:
A couple of changes that I'd like to see in a 2.0.0 release would be
-  the default declaration with no constructor parameters be to not call the Wire initialization, which is a change from the way it is in  1.2.2
which would allow declaring an object as
```
DS3232RTC myrtc;
```
vs 
```
DS3232RTC myrtc(0);
```
In fact you may want to consider dropping the ability to do any sort of h/w initialization in the constructor all together to avoid certain issues and ensure consistency across cores. Force the use of begin()

- change the .properties to indicate that other architectures other than AVR are supported.

bperrybap, 2018-04-07:
I get no compiler warnings with full warnings enabled from the DS3232RTC code on AVR, SAM, pic32 (chipkit), or ESP8266.
Using IDE 1.8.5 and the latest cores. (both pic32 and esp8266 have recent updates)
I modified and compiled the TimRTC example.  I declared an object named "myrtc" in TimeRTC sketch and modified the code accordingly to use it instead of "RTC".

I'm currently actually using the DS3232RTC library on esp8266 with the WeMos D1 mini and could easily test it out on a chipkit Uno32 if you would like; but I don' anticipate any issues.

Luckily the latest IDE will allow a sketch to still use the library on a core that the library does not explicitly support - but it will spit out a warning.
Iin my copy of the library I patched the architecture to be '*' to eliminate the warning.

JChristensen, 2018-04-07:
> Luckily the latest IDE will allow a sketch to still use the library on a core that the library does not explicitly support - but it will spit out a warning.

That's the warning I meant.

laboiteare, 2018-09-28:
Any news for this release ? Because, your library is very helpfull and I think ESP8266 deserves a fully compatible library like yours ;)

Thanks for all the work you do !

Pablito86c, 2018-10-29:
> Any news for this release ? Because, your library is very helpfull and I think ESP8266 deserves a fully compatible library like yours ;)
> 
> Thanks for all the work you do !

For the ESP8266 it would be great if the INT pin can wake up from the deep sleep. Something like a short pulse at the INT pin when alarm was triggered

JChristensen, 2018-10-29:
> For the ESP8266 it would be great if the INT pin can wake up from the deep sleep. Something like a short pulse at the INT pin when alarm was triggered

@Pablito86c, the library supports that. See the datasheet and also https://github.com/JChristensen/DS3232RTC/blob/master/alarm-primer.md

Pablito86c, 2018-10-29:
> > For the ESP8266 it would be great if the INT pin can wake up from the deep sleep. Something like a short pulse at the INT pin when alarm was triggered
> 
> @Pablito86c, the library supports that. See the datasheet and also https://github.com/JChristensen/DS3232RTC/blob/master/alarm-primer.md

I don't understand one thing. You say: "Once an alarm match occurs and the flag is set, the flag remains set until we tell the RTC to clear it" . If i tie the INT pin to the RST pin of my ESP8266 i can't wake up my ESP. In theory i should see a reset in my ESP, but nothing appens...

Thanks for your time

Best regards

Paolo

JChristensen, 2018-10-29:
Paolo, I prefer to use issues here strictly for problems with the library as opposed to debugging other folks' code and circuitry. But the first thing I would do is check the state of the INT pin (e.g. with a DVM) to ensure it's behaving as intended.

Good luck and best regards ... Jack

lmorris99, 2022-01-16:
Jack, this thread seems vaguely similar to my problem - didn't want to open a new Issue, not sure of github etiquette ...
Using the Megacorex arduino core, i get this:
`
Users/lindsay/Documents/Arduino/libraries/DS3232RTC-master/src/DS3232RTC.cpp:70:0: warning: **"RTC_STATUS" redefined**
 #define RTC_STATUS 0x0F
 
In file included from /Users/lindsay/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/avr/io.h:691:0,
                 from /Users/lindsay/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/avr/pgmspace.h:90,
                 from /Users/lindsay/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.9/cores/coreX-corefiles/api/String.h:30,
                 from /Users/lindsay/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.9/cores/coreX-corefiles/api/IPAddress.h:25,
                 from /Users/lindsay/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.9/cores/coreX-corefiles/api/Client.h:22,
                 from /Users/lindsay/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.9/cores/coreX-corefiles/api/ArduinoAPI.h:29,
                 from /Users/lindsay/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.9/cores/coreX-corefiles/Arduino.h:23,
                 from /Users/lindsay/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.9/libraries/Wire/src/Wire.h:25,
                 from /Users/lindsay/Documents/Arduino/libraries/DS3232RTC-master/src/DS3232RTC.cpp:31:
/Users/lindsay/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/**avr/iom4808.h:1979:0: note: this is the location of the previous definition
 #define RTC_STATUS  _SFR_MEM8(0x0141`**

Do you have any wisdom for me?  I appreciate your help.

JChristensen, 2022-01-16:
Hi Lindsay,

There's a lot going on in this thread which is why it's still hanging fire. I think your issue is somewhat narrower, basically a name collision between the DS3232RTC library and the MegaCoreX core.

I've opened a separate issue (#92). It should be straightforward but give me a few days. Basically sloppy coding on my part. There's a bunch of #defines that should probably be within the class definition so they don't cause conflicts in the global scope.

In the meantime, as a workaround, changing every occurrence of RTC_STATUS in the DS3232RTC.cpp file to something else should address the issue. Please try replacing the DS3232RTC.cpp file with [this one](https://gist.github.com/JChristensen/496533c1eb9ff03d3c8151ede1126560) and let me know if that helps.

Thanks ... Jack

JChristensen, 2022-02-09:
@bperrybap, I've pushed my [development branch](https://github.com/JChristensen/DS3232RTC/tree/dev) which is the release candidate for version 2.0.0. If you have time, please have a look, feedback welcome as always. I've added a [note in the README](https://github.com/JChristensen/DS3232RTC/tree/dev#important-note-library-v200) about backward compatibility with previous versions.

Thanks and best regards ... jc

bperrybap, 2022-02-18:
The library worked for my simple use which is as a TimeLib syncprovider.
I did have to make some changes to the code for both AVR and non AVR builds.
If I modified the .h file to have the added constructor, then the existing code worked "as is" for non AVR builds.

Part of what makes this a bit messy for code that wants to run on the old and new library is that there is no way for the sketch code to determine which version of library is begin used and hence which API is required.
Adding the dummy constructor for backward compatibility can at least alleviate the issue for the non AVR platforms.
Just something to consider.

For my hd44780 library I have defines for the library version.
But if you go that route you should create some automation scripts to use when doing releases that modify all various files that contain version information to ensure that they all get updated properly on easy release.
I have a script that takes a version number and goes and fixes several files like header files, README files, the properties file,  and documentation files and even commits the changes to ensure that everything is always in sync.

A simpler alternative might be to have a simple define in the header file that can indicate the major release number so you can tell if the library is a 1.x or 2.x library.
i.e. if the specific define exists and is greater than 1, then the code could compile adjust appropriately, particularly if on AVR platforms.
but that would be an advanced feature/capability that most users wouldn't use so it may not be worth it.
The only people that would use it would be people that want to provide code that can run on older and the newer library.
So it may not be worth it.

Another thing to consider while you are modifying the API is perhaps changing begin() to return a status as to whether or not it worked. i.e. a non zero return status indicates that the initialization in begin() didn't not complete successfully.

JChristensen, 2022-02-21:
Well it's been a little busier than usual here, so it took me a few days to get back to this. Great suggestions, thanks for those and for your time. A couple questions though.

I like the idea of the additional constructor for backwards compatibility but the `(void) initI2C;` part has me scratching my head. That looks like a C-style cast but I don't know what it does, or why the following would not work as well:
```c++
DS3232RTC(bool initI2C) {} // undocumented for backward compatibility
```
Second, as for `begin()` returning a status, I'm not sure. All `DS3232RTC::begin()` does is call `Wire.begin()` which does not return a status.

Thanks again, I've tested all the example sketches and also asked a couple other folks to try out 2.0.0. So I hope it's pretty solid at this point and I can push it soon.

bperrybap, 2022-02-21:
The void is to prevent a warning from an unused parameter

Good point about not needing to return a status value from begin() given that there is no status to return since Wire library doesn't return any status from its begin().

JChristensen, 2022-02-21:
I wondered about a warning but when I tried it I didn't see one. Maybe I don't have warnings turned up all the way.

bperrybap, 2022-02-21:
I'm guessing you don't have warnings enabled in the IDE.
I REALLY wish the IDE shipped with all the warnings enabled by default.
even better would be to ship with -Werror enabled to force people to not only see the warnings but actually break the compile to force people to fix their code as many times the warnings are an indication of an actual issue.

i.e. change the IDE so warnings are never hidden but the user could disable treating warnings as errors if he really wanted to.

JChristensen, 2022-02-21:
I had warnings set to "More." When I changed to "All" I do get the warning. Thanks!

JChristensen, 2022-02-22:
Closed by [07dd8f0](https://github.com/JChristensen/DS3232RTC/commit/07dd8f02bf8e807b5274b80914b19afc890f4ae2)

closed 2022-02-22 by JChristensen