JChristensen/DS3232RTC #51: Issue with including from a non-sketch file

Doridian, 2018-04-08:
When you #include the newest version of this library from a file that isn't a sketch (.ino file), such as a normal .cpp or .c file, the compiler complains that `byte` isn't defined, and therefor of course is unable to compile the project.
The fix right now is to include Arduino.h manually in your .cpp file before including your library, but I'd argue the include of Arduino.h should be left in the header to make it not dependant on inclusion order (or forcing everyone to include Arduino.h manually).

JChristensen, 2018-04-08:
Thanks for the catch. Fix coming soon.

closed 2018-04-08 by JChristensen

elmar1, 2018-04-13:
[image]

Works with time but the date is off.

elmar1, 2018-04-13:
That was with a Arduino UNO

JChristensen, 2018-04-13:
@elmar1: First, I don't see how I can know that the RTC was set correctly. Second, if there is an issue with the library, please create a new issue rather than making additional comments on an unrelated closed issue, and include a [SSCCE](http://sscce.org/) as text code that I can compile (an image of the code is less useful).
But, before doing that, please have a good look at how the RTC was set because the TimeRTC example sketch is known to function correctly, plus I don't have any other reports of similar issues.

commit f64080c6 (2018-04-08): Fixes #51.