JChristensen/DS3232RTC #53: SetSerial: 'RTC' not declared in this scope

philiporlando, 2018-04-25:
The SetSerial sketch fails to compile after including .ZIP library.
Using library DS3232RTC-master at version 1.2.0.

JChristensen, 2018-04-26:
For non-AVR architectures (e.g. ESP8266) a DS3232RTC object needs to be defined, so a slight modification to the example sketch would be needed. See https://github.com/JChristensen/DS3232RTC#constructor

closed 2018-04-26 by JChristensen

philiporlando, 2018-04-26:
Thanks for the response, and apologies for filing an issue that has already been addressed. However, I am still running into issues, and the ability to manually adjust the RTC without the lag-time from SetTime sketch is crucial for the success of my research project. 

Would you mind providing some additional information for getting the SetSerial sketch to work with non-AVR boards. I am using a Teensy3.2 Cortex-M4 architecture for this project. 

I have modified the SetSerial example to include the following, but I am running into a `SetSerial:49: error 'RTC' was not declared in this scope` error. 

Here's what my header and setup loop look like:

`
#include <DS3232RTC.h>        //http://github.com/JChristensen/DS3232RTC
#include <Streaming.h>        //http://arduiniana.org/libraries/streaming/

// For non-AVR boards only. Not needed for AVR boards.
DS3232RTC myRTC(false);     // tell constructor not to initialize the I2C bus.

void setup(void)
{
    Serial.begin(115200);
    myRTC.begin();          // initialize the I2C bus here.
    //setSyncProvider() causes the Time library to synchronize with the
    //external RTC by calling RTC.get() every five minutes by default.
    setSyncProvider(RTC.get);
    Serial << F("RTC Sync");
    if (timeStatus() != timeSet) Serial << F(" FAIL!");
    Serial << endl;
}
`

Your input is greatly appreciated!

philiporlando, 2018-04-26:
This did the trick. Clearly I have a lot more to learn...

A million thank yous!