JChristensen/DS3232RTC #55: What does this mean?

cyberlink1, 2018-06-18:
> To use this signal as an actual interrupt to a microcontroller, it will need to be connected properly and programmed in the application firmware. on the RTC.

Im reading through the docs and the examples (While I wait on a DS3232RTC in the mail) and am a bit confused by the statement.

Are you saying that you have to change the firmware on the RTC or are do you mean that you need to set the alarm and Square wave to NONE before you call  RTC.alarmInterrupt ?

So far, it looks great and looks to have all the functions I was looking for. Thank you

JChristensen, 2018-06-18:
Greetings, it means I have a typo! The words "on the RTC." should be deleted. I'll get that fixed, thanks. (There is no firmware that can be modified on the RTC.) I take a bit of exception to the pin on the RTC being referred to as an "interrupt" because it's really just a logic signal output. Whether the signal is used to generate an interrupt on (for example) a microcontroller is a separate issue. If an interrupt is to be generated, the RTC "interrupt" pin has to be connected to the MCU and the MCU firmware must properly configure the interrupt and also provide for handling it. Hope that helps.

closed 2018-06-18 by JChristensen

cyberlink1, 2018-06-18:
That helps a lot! Thank you

When the logic signal from the RTC is set will it remain high tell a new alarm is set in the RTC?

Im thinking with a Mosfet and a few minor parts I could have that power up an ESP8266 and when the ESP sets the next alarm it would loose power. None of this deep sleep stuff, it would be off. lol

JChristensen, 2018-06-18:
That sounds like it could work. The RTC's alarm flag (A1F or A2F) remains asserted until cleared (via the `alarm()` function.) If "interrupts" are also enabled (via the `alarmInterrupt()` function), then the INT/SQW pin will reflect the state of the alarm flag. I do not believe that just setting a new alarm time will reset the alarm flag.

JChristensen, 2018-06-18:
Also note that the INT signal from the RTC is active low. So it will normally be a high logic level, and will go low when an alarm occurs.

commit 30de58bd (2018-06-18): Fix typo in README file. Closes #55.