JChristensen/DS3232RTC #61: Error while compiling for any SAM microcontroller.

Microzod, 2018-10-15:
Hello.
Thank you for sharing this great library which I'm sure have saved a lot of people a lot of time.

However I cannot compile the example code called TimeRTC for the Arduino Due nor for the Zero, the last part of the DS3232RTC.h files contains a comment that at least makes it look like this was written to work on at least SAMD but I cannot get it to work.

Even though I do understand the individual words of the error message I cannot grasp what it actually means, I intended to post the complete error message but I cannot figure out how to format it in a way that doesn't look awfully taxing to read through so I have opted to cut out all the standard output I always see when I use the Due, if requested I can post the entire error message.
I have the latest Arduino IDE(1.8.7) so this should be easy to reproduced, just try and compile the TimeRTC example for any SAM microcontroller such as the Due(available through board manager).

Error message(first time posting on github):
`
	WARNING: library DS3232RTC-master claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (sam) architecture(s).
	Generating function prototypes...
	
	DS3232RTC-TimeAlarms-Time_serialSet_test:13:25: error: request for member 'get' in '1074666080u', which is of pointer type 'Rtc*' (maybe you meant to use '->' ?)
	
		setSyncProvider(RTC.get);   // the function to get the time from the RTC
	
							^
	
	Using library DS3232RTC-master at version 1.2.5 in folder: C:\Users\David\Documents\Arduino\libraries\DS3232RTC-master 
	Using library Time-master at version 1.5 in folder: C:\Users\David\Documents\Arduino\libraries\Time-master 
	Using library Wire at version 1.0 in folder: C:\Users\David\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\Wire 
	exit status 1
	request for member 'get' in '1074666080u', which is of pointer type 'Rtc*' (maybe you meant to use '->' ?)
`

JChristensen, 2018-10-15:
Usage is different for non-AVR architectures. This is documented [here](https://github.com/JChristensen/DS3232RTC#constructor) but I hope to make the API consistent someday when I have time. For the SAM platforms, the `RTC` object causes a conflict. So use the constructor to instantiate a DS3232RTC object that is not named `RTC` and then refer to that object as needed. Here is a modified version of the TimeRTC sketch that should work on the Due or the Zero. There are only two changes, one line added and one line modified.
```c++
// Arduino DS3232RTC Library
// https://github.com/JChristensen/DS3232RTC
//
// Example sketch illustrating Time library with Real Time Clock.
// This example is identical to the example provided with the Time Library,
// only the #include statement has been changed to include the DS3232RTC library.

#include <DS3232RTC.h>      // https://github.com/JChristensen/DS3232RTC

DS3232RTC myRTC;    // ADDED THIS LINE

void setup()
{
    Serial.begin(9600);
    setSyncProvider(myRTC.get);   // MODIFIED THIS LINE
    if(timeStatus() != timeSet)
        Serial.println("Unable to sync with the RTC");
    else
        Serial.println("RTC has set the system time");
}

void loop()
{
    digitalClockDisplay();
    delay(1000);
}

void digitalClockDisplay()
{
    // digital clock display of the time
    Serial.print(hour());
    printDigits(minute());
    printDigits(second());
    Serial.print(' ');
    Serial.print(day());
    Serial.print(' ');
    Serial.print(month());
    Serial.print(' ');
    Serial.print(year());
    Serial.println();
}

void printDigits(int digits)
{
    // utility function for digital clock display: prints preceding colon and leading 0
    Serial.print(':');
    if(digits < 10)
        Serial.print('0');
    Serial.print(digits);
}
```

Microzod, 2018-10-15:
Thanks for your answer, I had managed to forget to write that I have read everything that's there to read about DS3232RTC and I had tried instantiating a new DS3232RTC object but I thought that I should try the code you posted anyway and it worked.

What I apparently did wrong was to write:
`DS3232RTC myRTC();`
But then I get an error message talking about:
`error: request for member 'get' in 'myRTC', which is of non-class type 'DS3232RTC()'`

But it works when I write as you did:
`DS3232RTC myRTC;`

I am mainly a C guy thus far and I have no clue what difference the "()" makes but in any case it works now.

Finally I just want to say thank you so very much, you've made it so much more easy(and possible at all) to implement cool projects using the DS3231, my own DS3231 code was awkward to say the least.
I hope you'll have a very nice day and sorry for creating this in the end non-issue.

closed 2018-10-15 by Microzod

JChristensen, 2018-10-15:
No worries, glad you have it working. Thanks!