JChristensen/DS3232RTC #63: DS3231-Arduino Uno_LCM1602lcd

theOracle1984, 2019-01-29:
This is a great way work with the ds3231 and a Arduino Uno. I connected a lcm1602 lcd to the Arduino (connected to 0x3F) now in my code i need to write time to the lcd..

#include <DS3232RTC.h>   
#include <Time.h>
#include <Wire.h>
int val ;
//LCD add
#include <LCD.h>
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x3F, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE); // Addr, En, Rw, Rs, d4, d5, d6, d7, backlighpin, polarity

void setup()
{
  Serial.begin(9600);
  pinMode(7, OUTPUT); //relay1
  //LCD add
  lcd.begin(16,2);
    
  /////////Uncomment both lines below when you want to set the time
  /////////AFTER SETTING TIME COMMENT the lines.
  //setTime(16, 50, 00, 13, 01, 2019);//Hours / minutes / second / day of month / month / year
  //RTC.set(now());

  setSyncProvider(RTC.get);   // the function to get the time from the RTC
  if (timeStatus() != timeSet)
    Serial.println("Unable to sync with the RTC");
  else
    Serial.println("RTC has set the system time");
}

void loop()
{
  digitalClockDisplay();
  AutomaticProgram();
  
}

void AutomaticProgram()
{ //we are using hour as a value the rest is ignored.
  val = (hour());             //Bloei/Night - Bloei/Nacht between 19:00 and 07:00 aan, rest uit
  if ((19 == val))
  {
    digitalWrite(7, HIGH);
  }
  else if (val == 7)
  {
    digitalWrite(7, LOW);
  }
  else
  {
    digitalWrite(7, LOW);
  }
}

void digitalClockDisplay(void)
{
  // digital clock display of the time
  Serial.print(hour());
  printDigits(minute());
  printDigits(second());
  Serial.print(' ');
  Serial.print(day());
  Serial.print(' ');
  Serial.print(month());
  Serial.print(' ');
  Serial.print(year());
  Serial.println();
  lcd.backlight();
    lcd.setCursor(0, 0);
    lcd.print(RTC.readRTC);
    lcd.setCursor(0, 1);
    lcd.print(RTC.temperature());
    lcd.setCursor(12, 1);
    lcd.print("");
}

void printDigits(int digits)
{
  // utility function for digital clock display: prints preceding colon and leading 0
  Serial.print(':');
  if (digits < 10)
    Serial.print('0');
  Serial.print(digits);
}

i use this code to get the lcd to work.. 
the  lcd.print(RTC.readRTC); and   lcd.print(RTC.temperature()); 
dont really work. What should i put in the lcd.print( ) to print the time date year (and maybe temperature?)

JChristensen, 2019-01-29:
Example [here](https://github.com/JChristensen/DS3232RTC/blob/30de58bd0a673c3076ba151bdc6215ce8e0c87f4/examples/SetSerial/SetSerial.ino#L85-L87).

closed 2019-01-29 by JChristensen

theOracle1984, 2019-01-30:
Thanks that’s great. I already saw that yes.. But how does it work? How do I implement it into code?