JChristensen/DS3232RTC #64: SQW Alarm with samd21

maxholgasson, 2019-02-11:
Hi,

is there a way to use your lib with a Ardunio Zero?

I'm getting a lot of error e.g 
/var/folders/v7/k5b0gs_54q91l7p5r85snd_h0000gn/T/arduino_modified_sketch_362800/alarm_ex2.ino: In function 'void loop()':
alarm_ex2:71: error: request for member 'alarm' in '1073746944u', which is of pointer type 'Rtc*' (maybe you meant to use '->' ?)
         RTC.alarm(ALARM_1);    // reset the alarm flag
             ^
exit status 1

JChristensen, 2019-02-11:
See https://github.com/JChristensen/DS3232RTC#ds3232rtcbool-initi2c
Try instantiating your own DS3232RTC object and name it something other than "RTC".

closed 2019-02-11 by JChristensen

maxholgasson, 2019-02-14:
Thanks!

Do I need to init it with `myRTC.begin(); `after every wake up?
I put the mcu to sleep. The RTC is still supplied with power. After waking up do I need to execute `myRTC.begin();` again?

JChristensen, 2019-02-14:
`begin()` just initializes the I2C bus so once should do it. I have projects with AVR MCUs that sleep and only one call to `begin()` is needed. I haven't used the SAMD21; I suppose it _might_ be different in that regard, but I'd be a bit surprised if it were.