JChristensen/DS3232RTC #66: Negative temperatures & ARM-based Arduinos

adamgarbo, 2019-09-09:
Hi there,

I recently migrated to the Adafruit Feather ecosystem and discovered that negative temperatures do not appear to be read correctly from the DS3231 when using ARM-based microcontrollers. I did a quick test with an ARM and AVR processor to confirm:

**AVR - Adafruit Feather 32u4 + Adafruit DS3231 Featherwing**
10:49:30.739 -> t: 7
10:49:30.739 -> celsius: 1.75
10:49:35.758 -> t: -5
10:49:35.758 -> celsius: -1.25

**ARM - Adafruit Feather M0 + Adafruit DS3231 Featherwing**
10:53:48.093 -> t: 5
10:53:48.093 -> celsius: 1.25
10:53:53.126 -> t: 1023 _(transition to negative temperature)_
10:53:53.126 -> celsius: 255.75

I read through previous issues and it seems like this was an expected eventuality. Would it require considerable modification of the library to enable both AVR and ARM-based processors to be able to correctly read negative temperatures?

Cheers,
Adam

JChristensen, 2019-09-09:
Hello Adam,

I can have a look. Would you please provide the code you used to test with? Please pare it down as much as possible.

Thanks ... Jack

adamgarbo, 2019-09-09:
Hi Jack,

Thanks for the quick reply. I pulled a minimal amount of code from your GitHub examples to test, which I've included below:

```
#include <DS3232RTC.h>

DS3232RTC myRTC(false);

void setup()
{
  Serial.begin(115200);

  myRTC.begin();
  myRTC.setAlarm(ALM1_MATCH_DATE, 0, 0, 0, 1);
  myRTC.setAlarm(ALM2_MATCH_DATE, 0, 0, 0, 1);
  myRTC.alarm(ALARM_1);
  myRTC.alarm(ALARM_2);
  myRTC.alarmInterrupt(ALARM_1, false);
  myRTC.alarmInterrupt(ALARM_2, false);
  myRTC.squareWave(SQWAVE_NONE);

}

void loop()
{
  int t = myRTC.temperature();
  float celsius = t / 4.0;
  Serial.print("t: "); Serial.println(t); 
  Serial.print("celsius: "); Serial.println(celsius);
  delay(5000); 
}
```

I'd also be curious to know if the temperature values are rolling over in a predictable and repeatable fashion. As we have several instruments deployed in the field that are experiencing this issue, I'd be keen to try and retrieve the actual temperatures from the raw binary values.

Cheers,
Adam

JChristensen, 2019-09-09:
Adam, I just pushed version 1.2.7. Would you please test it on both platforms and let me know the results? Download it from GitHub or I think the Arduino Library Manager will pick it up within an hour.

I think the issue was that the `temperature()` function was defined as `int`, and `int` is 16 bits for AVR but 32 bits for ARM; this caused the sign bit to not be properly extended. So I suppose that it is predictable/repeatable. You could also read the RTC registers yourself, e.g. with the `readRTC()` function.

adamgarbo, 2019-09-09:
Thanks Jack!

I’ll test the changes this afternoon and report back. I appreciate the
quick fix!

Cheers,
Adam

adamgarbo, 2019-09-09:
Hi Jack,

I just finished testing the new 1.2.7 version and negative temperature readings are now working correctly for both the Adafruit Feather 32u4 and M0 boards. In the Arduino IDE, under examples, the DS3232RTC is still categorized as "INCOMPATIBLE", but otherwise working just fine!

Thanks again for your help,
Adam

closed 2019-09-09 by adamgarbo

JChristensen, 2019-09-09:
Hi Adam,

Thanks very much for testing. I don't have any ARM boards so that is why I have architectures=avr in the library.properties file. This issue reinforces my thinking in doing that. But, if the warning message bothers you, just change it to architectures=*

I was just testing with an Uno and it was looking OK using R-152a to create subzero temperatures (R-152a boils at -25°C).

Cheers ... Jack