JChristensen/DS3232RTC #67: I  cant get it to work with DUE

JPluess, 2019-09-24:
Hi there, 
it works with leonardo but not with due.
setSyncProvider(RTC.get); has trouble i do not understand the issue. can you run it on a due?

MultiButton:39:23: error: request for member 'get' in '1074666080u', which is of pointer type 'Rtc*' (maybe you meant to use '->' ?)

   setSyncProvider(RTC.get);

                       ^

F:\Projekte\Arduino\My Libraries\MultiButton\MultiButton V0.3\example\MultiButton\MultiButton.ino: In function 'void loop()':

MultiButton:70:11: error: request for member 'set' in '1074666080u', which is of pointer type 'Rtc*' (maybe you meant to use '->' ?)

       RTC.set(t);        // use the time_t value to ensure correct weekday is set

           ^

MultiButton:85:21: error: request for member 'temperature' in '1074666080u', which is of pointer type 'Rtc*' (maybe you meant to use '->' ?)

       float c = RTC.temperature() / 4.;

                     ^

exit status 1
request for member 'get' in '1074666080u', which is of pointer type 'Rtc*' (maybe you meant to use '->' ?)

JChristensen, 2019-09-24:
Hello,

For non-AVR architectures, it is necessary to [instantiate a DS3232RTC object](https://github.com/JChristensen/DS3232RTC#constructor).

Try the following test sketch and let me know how it works for you.

```c++
#include <DS3232RTC.h>      // https://github.com/JChristensen/DS3232RTC
DS3232RTC myRTC;

void setup()
{
    Serial.begin(9600);
    myRTC.begin();
    setSyncProvider(myRTC.get);   // the function to get the time from the RTC
    if(timeStatus() != timeSet)
        Serial.println("Unable to sync with the RTC");
    else
        Serial.println("RTC has set the system time");
}

void loop()
{
    digitalClockDisplay();
    delay(10000);
}

void digitalClockDisplay()
{
    // digital clock display of the time
    Serial.print("The current date and time is: ");
    Serial.print(hour());
    Serial.print(':');
    Serial.print(minute());
    Serial.print(':');
    Serial.print(second());
    Serial.print(' ');
    Serial.print(day());
    Serial.print('.');
    Serial.print(month());
    Serial.print('.');
    Serial.print(year());
    Serial.print(" - Temperatur: ");
    Serial.print(myRTC.temperature() / 4.0);
    Serial.print("°C");
    Serial.println();
}
```

JPluess, 2019-09-26:
thank you so much for your kind replay. 
it does compile and it does upload or so it seems. 
i get message:
WARNING: library DS3232RTC claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (sam) architecture(s).
Sketch uses 24204 bytes (4%) of program storage space. Maximum is 524288 bytes.
Atmel SMART device 0x285e0a60 found

did you have success with your due board?

I do not have any, 
creating the object will crash the controller. (DS3232RTC myRTC;)
no serial communication or led is blinking.

#include <DS3232RTC.h>      // https://github.com/JChristensen/DS3232RTC
DS3232RTC myRTC;

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin 13 as an output.
  pinMode(13, OUTPUT);

  Serial.begin(9600);
  //myRTC.begin();
  //setSyncProvider(myRTC.get);   // the function to get the time from the RTC
  //if (timeStatus() != timeSet)
	 // Serial.println("Unable to sync with the RTC");
  //else
	 // Serial.println("RTC has set the system time");
}

// the loop function runs over and over again forever
void loop() 
{
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
  Serial.print("The current date and time is: ");
  //digitalClockDisplay();
}

JChristensen, 2019-09-26:
Hi again,

I'm afraid I'm at a loss as I do not have a Due board. That is why the library is marked only for AVR architecture (and hence the warning message); my policy is to only indicate those architectures that I have personally tested and am in a position to troubleshoot. Still, I am aware that many of my libraries run fine on various architectures.

Seems pretty extreme that just defining the object causes such an issue. Just a shot in the dark, but if you instead define the object as:
`DS3232RTC myRTC(false);`
does that help at all?

JPluess, 2019-09-26:
hi dear, yes i got it to work. 

some libs use sda1 scl1 so i did not have any success there. 
i found the flase statement to be working, other wise the cpu would not run.

if you want i donate a due board to you from china. i am greatful for you help

i uploaded the sketch setserial, it would look like it set but since i had it on the other i2c it would not communicate to the ds3231 the serial would not show an issue. 

thanks so much. i feel relieved.

closed 2019-09-26 by JPluess