JChristensen/DS3232RTC #68: Conflict with SerialUSB on ARM chip

lmorris99, 2019-10-21:
Hi, Jack!  

Using DS3232RTC 1.2.9;  ATSAMD21G18 chip, Arduino IDE 1.8.10 using Board type MKRZERO: 
When I instantiate my DS3232RTC object, logging messages to SerialUSB stop.
Can you help / advise / offer breadcrumbs?

Details:
When I comment out line 3 below, Serial monitor says "test ... test ..." etc. Good.
Status LED stays dark. Good.
Serial port for MKRZERO stays alive. Good.

When I leave in line 3, ie, when I instantiate my DS3232RTC object, nothing happens on serial monitor. That's the problem.
I also notice that the Status LED on my (custom) PCB is slowly dimming and brightening.  Also, after uploading the script, the serial Port vanishes, and I have to click Reset button twice to make it come back.

Any idea what's going on?

#include<TimeLib.h>
#include <DS3232RTC.h>
DS3232RTC myrtc;
// time_t myTime;

void setup() {
  //myrtc.begin();
  SerialUSB.begin(115200);
  SerialUSB.println("begin...");

}

void loop() {
  //myTime=myrtc.get();
  //Serial1.println(second(myTime));
  SerialUSB.println("test");
  delay(1000);
}

JChristensen, 2019-10-21:
Hello Lindsay!
Wow that is a weird one. (Good job reducing the code down to a minimum!) When running that code, is the RTC actually connected?
Could this be a port multiplexing issue? Otherwise I don't know why I2C would interfere with Serial/USB.
Have you successfully used other I2C devices with the ATSAMD21G18? I don't have any of the ARM boards so I can't actually test it here.

lmorris99, 2019-10-21:
The RTC is indeed connected, soldered in in fact. This is a brand new PCB design, so there's some minor doubt about whether it's really connected or not but I'm 99% sure yes.
I have not successfully used any other I2C devices with this new PCB and ATSMD21 chip.
Perhaps it is indeed a port multiplexing issue or some sort of I2C vs Serial issue.  I'll chase tings in that direction. Thanks.

JChristensen, 2019-10-21:
I might also try one of the I2C scanners: https://playground.arduino.cc/Main/I2cScanner/

JChristensen, 2019-11-04:
@lmorris99 - any progress on this?

lmorris99, 2019-11-04:
My electrical engineer thinks that 
myrtc.begin() is not reaching the RTC, so it tries to communicate endlessly.

I Tried using i2cscanner, as you suggested. It would always hang.

Then I physically removed the RTC module, and it completed - It did not detect any devices, of course.

So maybe the problem is hardware and the RTC module rather than your library. 

I am trying to get it to detect a different i2c device, but I’m having trouble mapping physical pins on the Atsamd21g18 Chip to pin names in the arduino IDE. Looking at variants.cpp and variants.h, where is the definition for physical pin 12? Is it the 12th item in the PinDescription array? Hours of googling and I can’t turn pin 12 on. 

Any advice re pin mappings for Atsamd21g18 welcome ( I do have the data sheet but can’t figure it out. )

JChristensen, 2019-11-04:
Thanks for the update. I'll go ahead and close this issue then. If the scanner can detect the device, then this library should work as well. If the situation arises where the scanner detects the device but the library does _not_ work, then feel free to open another issue. Sorry but I'm clueless on the pin mappings/port multiplexing for the ATSAMD MCUs.

closed 2019-11-04 by JChristensen