JChristensen/DS3232RTC #74: Zero Arm SAMD

gioreva, 2020-02-25:
RTC Keyword already used on SAMD compiler.
Changing it with RTCb all work well.

gioreva, 2020-02-25:
No, not work.
When i add this line, all program not work.

  setSyncProvider(RTCb.get);   // the function to get the time from the RTC

JChristensen, 2020-02-25:
Does [this](https://github.com/JChristensen/DS3232RTC#description-2) help?

Description

For AVR architecture only (for backwards compatibility with the DS1307RTC library), the DS3232RTC library instantiates a DS3232RTC object named RTC; the user should not use the constructor for AVR boards. For other architectures, the user's code must instantiate a DS3232RTC object.

Syntax

DS3232RTC myRTC(initI2C);

gioreva, 2020-02-25:
For some problems, with ARM, RTC is initialized, and build is on error.
If i rename RTC with RTCb, compiler has no errors.
But if i add

DS3232RTC myRTC(initI2C);

exit status 1
'initI2C' was not declared in this scope

I believe that is compiling as AVR

JChristensen, 2020-02-25:
Please learn how to [read the README](https://github.com/JChristensen/DS3232RTC#parameters) :-D

"initI2C" is a boolean value, i.e. true or false. Or, it can be omitted and will default to true.

If "Arduino Zero" is selected (Tools > Board), it will compile for SAMD.

Please try the [rtcTimeTemp example sketch](https://github.com/JChristensen/DS3232RTC/blob/master/examples/rtcTimeTemp/rtcTimeTemp.ino), it compiles fine for either Arduino Uno or for Arduino Zero.

closed 2020-02-25 by JChristensen

gioreva, 2020-02-25:
No. do not build.
using original lib this is error
https://pastebin.com/SHDnHVAv

Renaming  RTC on RTCb this is the error.
https://pastebin.com/AzNGCD5q

gioreva, 2020-02-25:
By changing, in the example.
this
DS3232RTC myRTC;
with this
DS3232RTC myRTC(false);
the error is:
https://pastebin.com/89DmySVk

JChristensen, 2020-02-25:
I cannot duplicate the problem. I just did a fresh install of Arduino 1.8.12 on my test machine the other day and the rtcTimeTemp sketch compiles fine for both Uno and Zero.

From your first pastebin link, this error leads me to believe you have some other issues going on, maybe a damaged installation, etc. Note this is an **error** in the Arduino tools. I do not see this error on my machine.

`C:\Users\x-gio\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd21/include/samd21g18a.h:487:38: error: expected ')' before '*' token`

gioreva, 2020-02-25:
I have re downloaded library and now work.
Tomorrow I try on the hardware
After so?
  myRTC.begin();          // initialize the I2C bus
  setSyncProvider(myRTC.get);   // the function to get the time from the RT

The current library is so
#ifdef ARDUINO_ARCH_AVR
extern DS3232RTC RTC;
#endif

The library was put before is so
extern DS3232RTC RTC;

There are other version on git hub ?

JChristensen, 2020-02-25:
The latest version of the library is 1.2.10, available on GitHub or via the Arduino Library Manager.

gioreva, 2020-02-26:
Nothing, the board die.
Also serialUSB Dies.

#include <DS3232RTC.h>    //https://github.com/JChristensen/DS3232RTC
DS3232RTC myRTC;
myRTC.begin();          // initialize the I2C bus

gioreva, 2020-02-26:
I do not understand why Whire.begin() on the library crash the board.
And same Wire.begin() on startup schetch work well.

DS3232RTC::DS3232RTC(bool initI2C)
{
    if (initI2C) i2cBegin();
}

gioreva, 2020-02-26:
Re donwloading Time lib now work.
There are some wrong link on the Web.

But i do not undestandas how do I set the correct time, entering the local time and date.

time_t toUTC(time_t local);
myRTC.RTC.set(t);
ghour(local) = slc_ElemXKeyPadDataValGet(pGui, pvData);
myRTC.set(toUTC(local));
        

```
#include <Timezone.h>    // https://github.com/JChristensen/Timezone
// Central European Time (Frankfurt, Paris)
TimeChangeRule CEST = {"CEST", Last, Sun, Mar, 2, 120};     // Central European Summer Time
TimeChangeRule CET = {"CET ", Last, Sun, Oct, 3, 60};       // Central European Standard Time
Timezone CE(CEST, CET);
TimeChangeRule *tcr;        //pointer to the time change rule, use to get TZ abbrev

#include <TimeLib.h>      //https://github.com/PaulStoffregen/Time
#include <DS3232RTC.h>    //https://github.com/JChristensen/DS3232RTC
DS3232RTC myRTC;

  myRTC.begin();          // initialize the I2C bus
  setSyncProvider(myRTC.get);   // the function to get the time from the RTC

void loop()
{
  if( millis() - ClockScan > 1000 )
  {
        time_t utc = now();
        time_t local = CE.toLocal(utc, &tcr);
      
        snprintf_P(acTxt,20,"%02d:%02d:%02d",hour(local),minute(local),second(local));
        snprintf_P(acTxt,20,"%02d/%02d/%00004d",day(local),month(local),year(local));
```

gioreva, 2020-02-26:
Please, Help.
How do I set hardware clock, entering the local time.
I do not undestand time_t 
I have to read from clock, convert on locl time, change some, convert to UTC, and resend to clock

JChristensen, 2020-02-26:
I do not assist with user code problems here on GitHub. Please only raise issues here for actual problems with the library. Try the Arduino forum or other such venue for assistance with your code.