JChristensen/DS3232RTC #75: missreadings and lost RTC Time

spcqike, 2020-03-13:
Hello,

i have a problem, but i dont know if its related to the library or if all my DS's are broken(6 pcs).
i use DS3231 modules, which i thought are compatible to this library, with ESP32 boards.

my sketch uses the RTC.get() function to set the time and save the startup_time within setup{}, and updates the RTCtime once an hour over NTP (RTC.set()).
i noticed that my starttime is really odd, every once in a while. so i investigated and saw, that RTC.get() gives a negative value every now and than. after that, the RTCtime is way off.

for testing purposes i use 
`  uint32_t temp_z=0;
  while(RTC.get()>0){
    temp_z++;
  }
Serial.println(temp_z);`
to see how many RTC.get() it takes to get currupt data.
it takes from 5 to ~50.000 readings to get wrong times.

can anyone reproduce this or is it more a problem with my rtc modules?

thanks in advance

JChristensen, 2020-03-13:
Greetings,

I don't have an ESP32 so I cannot try to reproduce this. OTOH I am not aware of other similar reports.

Does `RTC.get()` ever return a zero value? This would indicate an error on the I2C bus. The code you give for testing purposes would mask such errors.

Other questions that come to mind: Where do your DS3231 modules come from? What value pullup resistors are used? What speed does the I2C bus run at? Do other I2C chips function correctly?

spcqike, 2020-03-13:
Hello,

RTC.get() returns a zero, but this happens very rarely (and I think this is related to the NTP update, which could happen at the same time)

the DS3231 modules come from AZ-Delivery (https://www.az-delivery.de/products/ds3231-real-time-clock?_pos=2&_sid=795fbbe56&_ss=r)
they seem to use 4,7kOhm as pullup. i don't know at what speed the I2C runs at. as for now, i just use RTC.begin(), .get() and .set().
i don't use other I2C devices in this project. but in other projects, some BME280/680 are working normaly.

in the meantime i testet another library, too. the problem also occurs there. 
i also noticed, that the librarys return diffent values after the missreading happened.

```
unix(DS3232RTC)   -   unix(DS3231)
1584102495 - 1584102495
[250x RTC.get() to provoke wrong reading]
1586989585 - 1587133585
```
it doesn't matter if i use RTC.get() from your DS3232RTC or RTC2.now().unixtime() from DS3231.h, both can corrupt the RTC internal time.
an update (RTC.set()) with NTP Time causes both libraries to read the same correct time again.

so, the problem need to be something with my setup (wrong pullup resistor, wrong i2c speed?) or the ds3231 module is cheap/crap...

JChristensen, 2020-03-13:
Yes it does sound like a hardware issue. I'm highly suspicious of those cheap modules, they cost significantly less than the chip itself, which is not an inexpensive chip IMHO. I've seen some that seem to work fine, but you never know. Counterfeit chips are a big problem, so you pays your money and you takes your chances. Personally I only use chips from Mouser or Digi-Key and I've never had issues. Good luck!

closed 2020-03-13 by JChristensen

spcqike, 2020-03-20:
one quick update about my issue,
it seems, that its related to the ESP32 and its dual core.
i2c communication seems to make weird things if you use booth cores, even if you just use 1 core for it and the other for other things.
right now, i stopped using both cores and it seems to work fine.