JChristensen/DS3232RTC #79: doesn't work on Nano Every

Blindekinder, 2020-06-01:
Output in french. Something like:
_pretends to be executable on avr architectures and could be incompatible with your actual board executing on megaavr_
any way to get it work there?

JChristensen, 2020-06-01:
This is only a warning. I only work with the AVR architecture. Since I cannot test or debug on other architectures, I mark my libraries as AVR only. Go ahead and try it anyway, it may work but I cannot guarantee. You can get rid of the warning by changing [this line in the library.properties file](https://github.com/JChristensen/DS3232RTC/blob/261ca7d0e62eea7902a84da93c4220138ee9f0b1/library.properties#L9) to read:
```c++
architectures=*
```
Cheers!

closed 2020-06-01 by JChristensen

Blindekinder, 2020-06-01:
Thank you!
but I get errors after that. In the TimeRTC example:
_TimeRTC:13:25: error: 'RTC_t {aka struct RTC_struct}' has no member named 'get'
     setSyncProvider(RTC.get);   // the function to get the time from the RTC
                         ^~~
exit status 1
'RTC_t {aka struct RTC_struct}' has no member named 'get'_

My script and the examples works well on Atmega328 based arduino

JChristensen, 2020-06-01:
Probably a DS3232RTC object needs to be instantiated [(see the README)](https://github.com/JChristensen/DS3232RTC#description-2).

Also try this sketch:
```c++
#include <DS3232RTC.h>      // https://github.com/JChristensen/DS3232RTC
DS3232RTC myRTC;

void setup()
{
    Serial.begin(9600);
    myRTC.begin();
    setSyncProvider(myRTC.get);   // the function to get the time from the RTC
    if(timeStatus() != timeSet)
        Serial.println("Unable to sync with the RTC");
    else
        Serial.println("RTC has set the system time");
}

void loop()
{
    digitalClockDisplay();
    delay(10000);
}

void digitalClockDisplay()
{
    // digital clock display of the time
    Serial.print("The current date and time is: ");
    Serial.print(hour());
    Serial.print(':');
    Serial.print(minute());
    Serial.print(':');
    Serial.print(second());
    Serial.print(' ');
    Serial.print(day());
    Serial.print('.');
    Serial.print(month());
    Serial.print('.');
    Serial.print(year());
    Serial.print(" - Temperatur: ");
    Serial.print(myRTC.temperature() / 4.0);
    Serial.print("°C");
    Serial.println();
}
```

Blindekinder, 2020-06-02:
yes, I already tried with _DS3232RTC myRTC;_ , but still doesn't work.

JChristensen, 2020-06-02:
I'm afraid that I'm out of ideas then. Sorry.