JChristensen/DS3232RTC #83: Running on mbed architecture(s)

tangtangdn, 2020-11-10:
Dear all,

I tried to use DS3231 as the external interrupt for Nano 33 BLE. But it cannot successfully compile. I tried the example of rtcTimeTemp which said should work on any architecture. The errors as shown as follows.

WARNING: library DS3232RTC claims to run on avr architecture(s) and may be incompatible with your current board which runs on mbed architecture(s).
In file included from c:\users\xxx\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\chrono:41:0,

                 from C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/mbed/rtos/Kernel.h:27,

                 from C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/mbed/rtos/Mutex.h:30,

                 from C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/mbed/platform/PlatformMutex.h:42,

                 from C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/mbed/drivers/AnalogIn.h:26,

                 from C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/Arduino.h:40,

                 from sketch\rtcTimeTemp.ino.cpp:1:

c:\users\xxx\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\ctime:64:11: error: '::clock' has not been declared

   using ::clock;

           ^~~~~

c:\users\xxx\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\ctime:65:11: error: '::difftime' has not been declared

   using ::difftime;

           ^~~~~~~~

c:\users\xxx\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\ctime:66:11: error: '::mktime' has not been declared

   using ::mktime;

           ^~~~~~

c:\users\xxx\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\ctime:67:11: error: '::time' has not been declared

   using ::time;

           ^~~~

c:\users\xxx\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\ctime:68:11: error: '::asctime' has not been declared

   using ::asctime;

           ^~~~~~~

c:\users\xxx\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\ctime:69:11: error: '::ctime' has not been declared

   using ::ctime;

           ^~~~~

c:\users\xxx\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\ctime:70:11: error: '::gmtime' has not been declared

   using ::gmtime;

           ^~~~~~

c:\users\xxx\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\ctime:71:11: error: '::localtime' has not been declared

   using ::localtime;

           ^~~~~~~~~

c:\users\xxx\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\ctime:72:11: error: '::strftime' has not been declared

   using ::strftime;

           ^~~~~~~~

exit status 1
Error compiling for board Arduino Nano 33 BLE.

Could someone help? Thank you very much.

JChristensen, 2020-11-10:
I get different errors, that seem to be related to PROGMEM variables in the Time library. I get the same errors when compiling the TimeRTC example sketch from the Time library. So I think you have something else going on, but once you resolve that, there are other issues too. Sorry I cannot be of more help.
```
WARNING: library DS3232RTC claims to run on avr architecture(s) and may be incompatible with your current board which runs on mbed architecture(s).
In file included from /home/jack/.arduino15/packages/arduino/hardware/mbed/1.3.0/cores/arduino/api/String.h:33:0,
                 from /home/jack/.arduino15/packages/arduino/hardware/mbed/1.3.0/cores/arduino/api/Print.h:24,
                 from /home/jack/.arduino15/packages/arduino/hardware/mbed/1.3.0/cores/arduino/api/Stream.h:25,
                 from /home/jack/.arduino15/packages/arduino/hardware/mbed/1.3.0/cores/arduino/api/Client.h:22,
                 from /home/jack/.arduino15/packages/arduino/hardware/mbed/1.3.0/cores/arduino/api/ArduinoAPI.h:29,
                 from /home/jack/.arduino15/packages/arduino/hardware/mbed/1.3.0/cores/arduino/Arduino.h:51,
                 from /home/jack/Arduino/libraries/Time/DateStrings.cpp:12:
/home/jack/Arduino/libraries/Time/DateStrings.cpp: In function 'char* monthStr(uint8_t)':
/home/jack/.arduino15/packages/arduino/hardware/mbed/1.3.0/cores/arduino/api/deprecated-avr-comp/avr/pgmspace.h:106:49: error: 'const void*' is not a pointer-to-object type
 #define pgm_read_ptr(addr) (*(const void *)(addr))
                                                 ^
/home/jack/.arduino15/packages/arduino/hardware/mbed/1.3.0/cores/arduino/api/deprecated-avr-comp/avr/pgmspace.h:63:45: note: in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src) strcpy((dest), (src))
                                             ^~~
/home/jack/Arduino/libraries/Time/DateStrings.cpp:72:29: note: in expansion of macro 'pgm_read_ptr'
     strcpy_P(buffer, (PGM_P)pgm_read_ptr(&(monthNames_P[month])));
                             ^
/home/jack/Arduino/libraries/Time/DateStrings.cpp: In function 'char* dayStr(uint8_t)':
/home/jack/.arduino15/packages/arduino/hardware/mbed/1.3.0/cores/arduino/api/deprecated-avr-comp/avr/pgmspace.h:106:49: error: 'const void*' is not a pointer-to-object type
 #define pgm_read_ptr(addr) (*(const void *)(addr))
                                                 ^
/home/jack/.arduino15/packages/arduino/hardware/mbed/1.3.0/cores/arduino/api/deprecated-avr-comp/avr/pgmspace.h:63:45: note: in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src) strcpy((dest), (src))
                                             ^~~
/home/jack/Arduino/libraries/Time/DateStrings.cpp:86:28: note: in expansion of macro 'pgm_read_ptr'
    strcpy_P(buffer, (PGM_P)pgm_read_ptr(&(dayNames_P[day])));
                            ^
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

closed 2020-11-10 by JChristensen

tangtangdn, 2020-11-10:
> I get different errors, that seem to be related to PROGMEM variables in the Time library. I get the same errors when compiling the TimeRTC example sketch from the Time library. So I think you have [...]

Dear Jack, thank you so much for your contributions and prompt responses to my questions.