JChristensen/DS3232RTC #84: Allarm every 2 hours

gpalmia, 2020-12-14:
Hello,
Can anyone suggest me how to create alarm every 2 hours with ds3231?
Thanks

JChristensen, 2020-12-14:
https://github.com/JChristensen/DS3232RTC/blob/master/alarm-primer.md

closed 2020-12-14 by JChristensen

sanastasiou89, 2021-01-07:
Hi,

I have a question in regards to the above link you have posted.

As per point 3 on the post, when a flag is set, it remains like this until we clear the flag. As per my understanding, this means that INT pin (we assume that we disable SQW through RTC.squareWave(SQWAVE_NONE))  remains active low until we clear the flag. Correct? So we can control the duration of an active low state through arduino delay function.

Thanks in advance.

JChristensen, 2021-01-07:
You are correct, but the RTC controls the state of its INT pin, not the `delay()` function.

sanastasiou89, 2021-01-07:
Thank you very much for your answer. So, how do i implement through code for example to keep the INT pin in active low state for 10 seconds and then set it high again (which means deactivation of alarm flag)?

JChristensen, 2021-01-07:
I try to limit issues here to actual problems with the library. If you need help with your code, please use the Arduino forum or one of several others that are available. Study the examples and the alarm primer that come with this library. Study the DS3231 datasheet. Sorry about that and good luck with your project.