JChristensen/DS3232RTC #85: Library for esp32

marcelocampilima, 2021-01-02:
Hi,
I am using this library for an Arduino Project. However, we upgraded this project for the esp32 microcontroller. 
When I was refactoring the code for esp32, I figured out that this library works only for Arduino platform. 
Is there a way to use it also for esp32? Because if it does not, I will need to refactor a lot of code already tested.

Thank you much for the library and for your excellent job !!

Best regards

JChristensen, 2021-01-02:
Hello, have you actually tried it on ESP32? I wonder if it may in fact work. Was there an error? Do not be put off by the compiler warning. But I do not use ESP32 so I cannot say for sure. I am told that it works on ESP8266, see Issue #78.

JChristensen, 2021-01-10:
Nothing heard, closing issue.

closed 2021-01-10 by JChristensen

marcelocampilima, 2021-01-11:
Hi,

Regardless of the warning message, it seems to be working so far.

Thank you so much for your attention to this matter!

Best regards,
Marcelo Lima