JChristensen/DS3232RTC #86: RTC not accurate time

Geeker3D, 2021-01-16:
Just completed making Lazy 7 One clock published by parallyze . Uses DS3232RTC library. Seems that the clock runs to fast. Is there any paramaters that the user can change that it would change the speed of the clock itself.

JChristensen, 2021-01-16:
Normally DS3231/32 are very accurate. See the datasheet; there is an Aging Offset register that can compensate about ±10ppm or so, but I've never had a reason to use it. More than likely the issue is either bad code or a bad part (or more likely a counterfeit part.) Authentic DS3231s are currently going for $9-10USD in single quantities. If it cost less than that, then it's probably not an authentic part and I wouldn't be surprised if it didn't operate to spec.

closed 2021-01-16 by JChristensen

Geeker3D, 2021-01-16:
Thank you Jack. You mentioned it might not be authentic.  Your probably
right. Ill reorder and try another. Thanks again