JChristensen/DS3232RTC #87: Ошибка с RTC, как исправить?(Arduino)

Membrum, 2021-02-17:
Arduino: 1.8.14 Hourly Build 2021/01/29 11:33 (Windows 10), Плата:"Arduino Uno"

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino: In function 'void loop()':

sketch_feb17a:103:12: error: 'RTC' was not declared in this scope

 int Hour = RTC.get(DS1307_HR,true); // Получаем значение текущего часа

            ^~~

sketch_feb17a:103:20: error: 'DS1307_HR' was not declared in this scope

 int Hour = RTC.get(DS1307_HR,true); // Получаем значение текущего часа

                    ^~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:103:20: note: suggested alternative: 'DS1307_ON'

 int Hour = RTC.get(DS1307_HR,true); // Получаем значение текущего часа

                    ^~~~~~~~~

                    DS1307_ON

sketch_feb17a:104:22: error: 'DS1307_MIN' was not declared in this scope

 int Minute = RTC.get(DS1307_MIN,false);

                      ^~~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:104:22: note: suggested alternative: 'DS1307_ON'

 int Minute = RTC.get(DS1307_MIN,false);

                      ^~~~~~~~~~

                      DS1307_ON

sketch_feb17a:105:22: error: 'DS1307_SEC' was not declared in this scope

 int Second = RTC.get(DS1307_SEC,false);

                      ^~~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:105:22: note: suggested alternative: 'DS1307_ON'

 int Second = RTC.get(DS1307_SEC,false);

                      ^~~~~~~~~~

                      DS1307_ON

sketch_feb17a:106:20: error: 'DS1307_DATE' was not declared in this scope

 int Date = RTC.get(DS1307_DATE,false);

                    ^~~~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:106:20: note: suggested alternative: 'DS1307_RAM'

 int Date = RTC.get(DS1307_DATE,false);

                    ^~~~~~~~~~~

                    DS1307_RAM

sketch_feb17a:107:19: error: 'DS1307_MTH' was not declared in this scope

 int Mon = RTC.get(DS1307_MTH,false);

                   ^~~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:107:19: note: suggested alternative: 'DS1307_ON'

 int Mon = RTC.get(DS1307_MTH,false);

                   ^~~~~~~~~~

                   DS1307_ON

sketch_feb17a:108:18: error: 'DS1307_YR' was not declared in this scope

 int Ye = RTC.get(DS1307_YR,false);

                  ^~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:108:18: note: suggested alternative: 'DS1307_ON'

 int Ye = RTC.get(DS1307_YR,false);

                  ^~~~~~~~~

                  DS1307_ON

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino: In function 'void SetTime()':

sketch_feb17a:154:1: error: 'RTC' was not declared in this scope

 RTC.stop();

 ^~~

sketch_feb17a:155:9: error: 'DS1307_SEC' was not declared in this scope

 RTC.set(DS1307_SEC,0); //set the seconds

         ^~~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:155:9: note: suggested alternative: 'DS1307_ON'

 RTC.set(DS1307_SEC,0); //set the seconds

         ^~~~~~~~~~

         DS1307_ON

sketch_feb17a:156:9: error: 'DS1307_MIN' was not declared in this scope

 RTC.set(DS1307_MIN,m); //set the minutes

         ^~~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:156:9: note: suggested alternative: 'DS1307_ON'

 RTC.set(DS1307_MIN,m); //set the minutes

         ^~~~~~~~~~

         DS1307_ON

sketch_feb17a:157:9: error: 'DS1307_HR' was not declared in this scope

 RTC.set(DS1307_HR,h); //set the hours

         ^~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:157:9: note: suggested alternative: 'DS1307_ON'

 RTC.set(DS1307_HR,h); //set the hours

         ^~~~~~~~~

         DS1307_ON

sketch_feb17a:158:9: error: 'DS1307_DATE' was not declared in this scope

 RTC.set(DS1307_DATE,D); //set the date

         ^~~~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:158:9: note: suggested alternative: 'DS1307_RAM'

 RTC.set(DS1307_DATE,D); //set the date

         ^~~~~~~~~~~

         DS1307_RAM

sketch_feb17a:159:9: error: 'DS1307_MTH' was not declared in this scope

 RTC.set(DS1307_MTH,M); //set the month

         ^~~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:159:9: note: suggested alternative: 'DS1307_ON'

 RTC.set(DS1307_MTH,M); //set the month

         ^~~~~~~~~~

         DS1307_ON

sketch_feb17a:160:9: error: 'DS1307_YR' was not declared in this scope

 RTC.set(DS1307_YR,Y); //set the year

         ^~~~~~~~~

C:\Users\Admin\Documents\Arduino\sketch_feb17a\sketch_feb17a.ino:160:9: note: suggested alternative: 'DS1307_ON'

 RTC.set(DS1307_YR,Y); //set the year

         ^~~~~~~~~

         DS1307_ON

Несколько библиотек найдено для "SPI.h"

 Используется: C:\Temp\Rar$EXa33596.29173\arduino-nightly\hardware\arduino\avr\libraries\SPI

Не используется: C:\Users\Admin\Documents\Arduino\libraries\SPI-master

exit status 1

'RTC' was not declared in this scope

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

JChristensen, 2021-02-17:
I can't do much with that. Does [the TimeRTC example sketch](https://github.com/JChristensen/DS3232RTC/blob/master/examples/TimeRTC/TimeRTC.ino) compile OK?

JChristensen, 2021-02-27:
Nothing heard. Closing issue.

closed 2021-02-27 by JChristensen