JChristensen/DS3232RTC #88: Exception when trying to write date/time

marcelocampilima, 2021-03-04:
Hello,

I am using the library  DS3232RTC on a project based on ESP32. It is happening something that I tried to figure out but I have not been able to.

When I try to update the Date using "write(tmElements_t)" I get an exception and the system restarts. One thing that is really strange to me is the date when "read(tmElements_t)" before writing.

Below you can see the code and after the debug of the data. 

// set the current date
void setCurrentDate(){
  tmElements_t currentDateTime; 

  byte currentDay = 4;
  byte currentMonth = 3;
  byte currentYear = 2021 - 1970;
  byte currentWeekDay = 5; 
  
  rtcClock.read(currentDateTime); 
  Serial.print("Date before updating: ");
  Serial.print(currentDateTime.Month, DEC);
  Serial.print("/");
  Serial.print(currentDateTime.Day, DEC);
  Serial.print("/");
  Serial.println(currentDateTime.Year + 1970, DEC);
  Serial.print("Weekday: ");
  Serial.println(currentDateTime.Wday, DEC);

  // Set the date
  currentDateTime.Day = currentDay;
  currentDateTime.Month = currentMonth;
  currentDateTime.Year = currentYear;
  currentDateTime.Wday = currentWeekDay;

  Serial.print("Date after setup: ");
  Serial.print(currentDateTime.Month, DEC);
  Serial.print("/");
  Serial.print(currentDateTime.Day, DEC);
  Serial.print("/");
  Serial.println(currentDateTime.Year + 1970, DEC);
  Serial.print("Weekday: ");
  Serial.println(currentDateTime.Wday, DEC);

  // writing the date to RTC
  rtcClock.write(currentDateTime);

  rtcClock.read(currentDateTime); 
  Serial.print("Date after updating: ");
  Serial.print(currentDateTime.Month, DEC);
  Serial.print("/");
  Serial.print(currentDateTime.Day, DEC);
  Serial.print("/");
  Serial.println(currentDateTime.Year + 1970, DEC);
  Serial.print("Weekday: ");
  Serial.println(currentDateTime.Wday, DEC);
}

Date before updating: 252/98/2033
Weekday: 240
Date after setup: 3/4/2021
Weekday: 5
Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d961f  PS      : 0x00060630  A0      : 0x800d99ba  A1      : 0x3ffcfce0  
A2      : 0x00000133  A3      : 0x00000000  A4      : 0x00000032  A5      : 0x3ffbedb8  
A6      : 0x00004e20  A7      : 0x60013000  A8      : 0x00000113  A9      : 0x3ffcfcc0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffe17b4  A13     : 0x00000001  
A14     : 0x00002710  A15     : 0x3ffbedb8  SAR     : 0x00000013  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400d961f:0x3ffcfce0 0x400d99b7:0x3ffcfd10 0x400d3a5d:0x3ffcfd30 0x400d3ab1:0x3ffcfd50 0x400d3ba5:0x3ffcfd70 0x400d3e56:0x3ffcfd90 0x400d3e91:0x3ffcfdb0 0x400d3f74:0x3ffcfde0 0x400d2543:0x3ffcfe00 0x400d27f7:0x3ffcfe30 0x400d1c82:0x3ffcfe60 0x400d1d67

Thanks in advance.

JChristensen, 2021-03-04:
I cannot duplicate the issue. I do not have an ESP32, but the code below appears to run fine on an Uno. Sorry.
```c++
#include <DS3232RTC.h>      // https://github.com/JChristensen/DS3232RTC

DS3232RTC rtcClock;

void setup()
{
    Serial.begin(115200);
    rtcClock.begin();
    setCurrentDate();
}

void loop()
{
}

// set the current date
void setCurrentDate()
{
    tmElements_t currentDateTime;

    byte currentDay = 4;
    byte currentMonth = 3;
    byte currentYear = 2021 - 1970;
    byte currentWeekDay = 5;

    rtcClock.read(currentDateTime);
    Serial.print("Date before updating: ");
    Serial.print(currentDateTime.Month, DEC);
    Serial.print("/");
    Serial.print(currentDateTime.Day, DEC);
    Serial.print("/");
    Serial.println(currentDateTime.Year + 1970, DEC);
    Serial.print("Weekday: ");
    Serial.println(currentDateTime.Wday, DEC);

    // Set the date
    currentDateTime.Day = currentDay;
    currentDateTime.Month = currentMonth;
    currentDateTime.Year = currentYear;
    currentDateTime.Wday = currentWeekDay;

    Serial.print("Date after setup: ");
    Serial.print(currentDateTime.Month, DEC);
    Serial.print("/");
    Serial.print(currentDateTime.Day, DEC);
    Serial.print("/");
    Serial.println(currentDateTime.Year + 1970, DEC);
    Serial.print("Weekday: ");
    Serial.println(currentDateTime.Wday, DEC);

    // writing the date to RTC
    rtcClock.write(currentDateTime);

    rtcClock.read(currentDateTime);
    Serial.print("Date after updating: ");
    Serial.print(currentDateTime.Month, DEC);
    Serial.print("/");
    Serial.print(currentDateTime.Day, DEC);
    Serial.print("/");
    Serial.println(currentDateTime.Year + 1970, DEC);
    Serial.print("Weekday: ");
    Serial.println(currentDateTime.Wday, DEC);
}
```
Output:
```
Date before updating: 1/1/2000
Weekday: 1
Date after setup: 3/4/2021
Weekday: 5
Date after updating: 3/4/2021
Weekday: 5
```

closed 2021-03-04 by JChristensen

marcelocampilima, 2021-03-04:
Hi,

Thank you for your reply.

Could you see these numbers when reading from RTC? 
Date before updating: 252/98/2033
Weekday: 240

What do you think this can be?

Best regards

JChristensen, 2021-03-04:
All the numbers looked good. See my output above. I have no idea what might be going on, sorry.

marcelocampilima, 2021-03-04:
ok. Thank you, anyway.

Best regards.