JChristensen/DS3232RTC #89: TimeRTC example

roblatour, 2021-04-05:
The TimeRTC example is missing the line:

JChristensen, 2021-04-05:
It compiles OK for an Uno. What board are you using? See the [documentation for the constructor](https://github.com/JChristensen/DS3232RTC#ds3232rtcbool-initi2c). Defining a DS3232RTC object for other architectures is left as an exercise for the user ;-)

roblatour, 2021-04-05:
An esp32 devkit 1

JChristensen, 2021-04-05:
Working as advertised then, I guess. While the example could be modified to include an instantiation of a DS3232RTC object, this is not without (a perhaps small) risk. If someone just copies the example, they could end up with an extra, unneeded object. Or the example could be modified with conditional compilation. Pros and cons to all of this, but overall I think it's a small issue. Maybe I will revisit in the future.

closed 2021-04-05 by JChristensen