JChristensen/DS3232RTC #93: Alarm doesn't work after 40 minutes

hoseinaghajari, 2022-01-17:
Hi, This is useful Library. thanks to you dear @JChristensen .
I have a problem with my code: I set the alarm for every 20 minutes. It trigger the INT0 of Arduino at first time but in second time (2*20=40minutes later) it doesn't Trig the INT0 and my Arduino never wakes up until I reset the device.
here is my code:

`
#include "LowPower.h"
#include <DS3232RTC.h> 
#include "DHT.h"        
#include <SPI.h>       
#include <SD.h>        

const int wakeUpPin = 2;
#define DHTPIN 4  
#define DHTTYPE DHT11  
DHT dht(DHTPIN, DHTTYPE);
File myFile;
const int chipSelect = 10;
const int time_interval=20;// Sets the wakeup intervall in minutes

void wakeUp()
{
     detachInterrupt(0); 
}

void setup() {
  Serial.begin(9600);
  pinMode(LED_BUILTIN,OUTPUT);
  pinMode(wakeUpPin, INPUT_PULLUP); 
  digitalWrite(LED_BUILTIN,HIGH);
  
    RTC.setAlarm(ALM1_MATCH_DATE, 0, 0, 0, 1);
    RTC.setAlarm(ALM2_MATCH_DATE, 0, 0, 0, 1);
    RTC.alarm(ALARM_1);
    RTC.alarm(ALARM_2);
    RTC.alarmInterrupt(ALARM_1, false);
    RTC.alarmInterrupt(ALARM_2, false);
    RTC.squareWave(SQWAVE_NONE);
   
     time_t t; //create a temporary time variable so we can set the time and read the time from the RTC
    t=RTC.get();//Gets the current time of the RTC
    RTC.setAlarm(ALM1_MATCH_MINUTES , 0, minute(t)+time_interval, 0, 0);// Setting alarm 1 to go off 5 minutes from now
    // clear the alarm flag
    RTC.alarm(ALARM_1);
    // configure the INT/SQW pin for "interrupt" operation (disable square wave output)
    RTC.squareWave(SQWAVE_NONE);
    // enable interrupt output for Alarm 1
    RTC.alarmInterrupt(ALARM_1, true);
    
    dht.begin();//Start the DHT sensor
 
    Serial.print("Initializing SD card...");
    if (!SD.begin(chipSelect)) {
        Serial.println("initialization failed!");
        return;
    }
    Serial.println("initialization done.");
}

void loop() {
 delay(5000);
 Going_To_Sleep();
}

void Going_To_Sleep(){
   attachInterrupt(0, wakeUp, LOW);
    digitalWrite(LED_BUILTIN,LOW);          
    time_t t;                                             
    t=RTC.get();                                             
    Serial.println("Sleep  Time: "+String(hour(t))+":"+String(minute(t))+":"+String(second(t)));
    delay(1000);                                       
     LowPower.powerDown(SLEEP_FOREVER, ADC_OFF, BOD_OFF); 
    Serial.println("just woke up!");
    digitalWrite(LED_BUILTIN,HIGH);
    temp_Humi();
    t=RTC.get();
    Serial.println("WakeUp Time: "+String(hour(t))+":"+String(minute(t))+":"+String(second(t)));

    RTC.setAlarm(ALM1_MATCH_MINUTES , 0, minute(t)+time_interval, 0, 0);
  
  RTC.alarm(ALARM_1);
  }

void temp_Humi(){

  float h = dht.readHumidity();
  float t = dht.readTemperature();
  float f = dht.readTemperature(true);
  writeData(h,t,f);
  Serial.print("Humidity: ");
  Serial.print(h);
  Serial.print(" %\t");
  Serial.print("Temperature: ");
  Serial.print(t);
  Serial.print(" *C ");
  Serial.print(f);
  Serial.print(" *F\t");
}

  void writeData(float h,float t,float f){
  time_t p;
  p=RTC.get();
  String file_Name=String(day(p))+monthShortStr(month(p))+String(year(p))+".txt";
  myFile = SD.open(file_Name, FILE_WRITE);
  if (myFile) {
    Serial.print("Writing to "+ file_Name);
    myFile.println(String(hour(p))+":"+String(minute(p))+" Hum: "+String(h)+"% C: "+String(t)+" F: "+String(f));
    myFile.close();
    Serial.println("done.");
  } else {
    Serial.println("error opening "+ file_Name);
  }
  myFile = SD.open(file_Name);
  if (myFile) {
    Serial.println(file_Name);
    while (myFile.available()) {
      Serial.write(myFile.read());
    }
    myFile.close();
  } else {
    Serial.println("error opening"+ file_Name);
  }
 }
`

JChristensen, 2022-01-17:
Greetings, @hoseinaghajari 

Sorry but I prefer to limit issues on GitHub to actual problems with the library, as opposed to issues with user code. Please consider using the Arduino forum or other similar venue.

Best regards and good luck.

closed 2022-01-17 by JChristensen

JChristensen, 2022-01-18:
I got to thinking this might make a good example sketch, so I put one together as a Gist. [Check it out here](https://gist.github.com/JChristensen/99306c19b28176930079eb8a968c505d) and I'll include it in the next release of the library.
Cheers!