JChristensen/DS3232RTC #94: Set Alarm for every 20 minutes

hoseinaghajari, 2022-01-18:
HI DEAR @JChristensen .

You closed my previous issue so my question is can we Set Alarm for every 20 minutes? How?

JChristensen, 2022-01-18:
[As I said](https://github.com/JChristensen/DS3232RTC/issues/93#issuecomment-1014909150), my policy is to only entertain issues here on GitHub that are actual problems with the library.

[But I also said](https://github.com/JChristensen/DS3232RTC/issues/93#issuecomment-1015018924) that I had written [an example sketch](https://gist.github.com/JChristensen/99306c19b28176930079eb8a968c505d) that illustrates the technique.

I hope you find it instructive. Please check it out.

Best regards.

closed 2022-01-18 by JChristensen

JChristensen, 2022-01-18:
By the way, I had another look at your sketch and I think I see an issue.
Would you like a hint or would you prefer to have a go at it yourself?

hoseinaghajari, 2022-01-18:
> By the way, I had another look at your sketch and I think I see an issue. Would you like a hint or would you prefer to have a go at it yourself?

I would like to wake my Arduino whenever I want. for instance; in one project every 20 min or in another project every one hour. I want to have flexibility for time intervals.

by the way:
I tested [that example](https://gist.github.com/JChristensen/99306c19b28176930079eb8a968c505d) but If I change the  **wakeInterval {5};** to **wakeInterval {10};** it will trig but won't go in sleep after that.
here is my output:
[image]

JChristensen, 2022-01-18:
I've updated the [example](https://gist.github.com/JChristensen/99306c19b28176930079eb8a968c505d); give it another try please.
A small delay is needed after waking up to allow communication to the RTC.

hoseinaghajari, 2022-01-19:
> I've updated the [example](https://gist.github.com/JChristensen/99306c19b28176930079eb8a968c505d); give it another try please. A small delay is needed after waking up to allow communication to the RTC.

I solved That. just by setting The RTD time before setting new alarm.
**But I have another question**. Due to I need D2(INT0) on Arduino UNO for another purpose, Can we define Wakeup pin into other pins? such as 3. 
I've actually changed the **const int wakeUpPin = 2;** to **const int wakeUpPin = 3;** but it didn't trig the pin.

JChristensen, 2022-01-19:
Pin 3 (INT1) should work the same, but note that the example sketch uses pin 3 to power the RTC. So that would need to be changed too.

JChristensen, 2022-01-19:
Opps, my bad. The `goToSleep()` function was always setting INT0. One more change to the example sketch. It should now work for either INT0/Pin2 or INT1/Pin3. Give it a try please. Thanks for finding that.

hoseinaghajari, 2022-01-19:
> Opps, my bad. The `goToSleep()` function was always setting INT0. One more change to the example sketch. It should now work for either INT0/Pin2 or INT1/Pin3. Give it a try please. Thanks for finding that.

**You changed STH in Code or in library files?**

JChristensen, 2022-01-19:
> You Solved it so thank you BUT I come to think it has new problem with waking up. **please tell me how did you solve `goToSleep()` function.**

It seems to be working OK here. What new problem do you think it has? I can't help unless you describe it.

It was a change to the example sketch. See [here](https://gist.github.com/JChristensen/99306c19b28176930079eb8a968c505d/revisions).

hoseinaghajari, 2022-01-19:
> It seems to be working OK here. What new problem do you think it has? I can't help unless you describe it.
>
> It was a change to the example sketch. See [...]

my connection is correct and here is my output:
I reset it 3 times but here it is:

[image]

hoseinaghajari, 2022-01-19:
> Opps, my bad. The `goToSleep()` function was always setting INT0. One more change to the example sketch. It should now work for either INT0/Pin2 or INT1/Pin3. Give it a try please. Thanks for finding that.

In previous sketch you used `LowPower.h `Library but now you use `avr `Library. working with `LowPower.h` is easy but it just available on INT0.

JChristensen, 2022-01-19:
> my connection is correct and here is my output:
> I reset it 3 times but here it is:

That looks like an I2C communication failure.
Try increasing the `delay(20);` on [line 89](https://gist.github.com/JChristensen/99306c19b28176930079eb8a968c505d#file-alarm_ex10-ino-L89) to `delay(100);`

JChristensen, 2022-01-19:
> In previous sketch you used `LowPower.h `Library but now you use `avr `Library. working with `LowPower.h` is easy but it just available on INT0.

<avr/sleep.h> is just for sleep functions, it doesn't really have anything to do with what interrupt is used to wake the MCU.

JChristensen, 2022-01-19:
OK, I've made one last revision to the [example sketch](https://gist.github.com/JChristensen/99306c19b28176930079eb8a968c505d), just to add some informational print statements.

All seems to be working so I think the example sketch is complete.

```
/home/jack/Dropbox/sketchbook/temp/alarm_ex10/alarm_ex10.ino Jan 19 2022 12:55:22
Sketch start 17:57:28 Wed 19 Jan 2022
Wake interval: 5 minutes.
Using interrupt INT0
RTC interrupt pin: 2
RTC power pin: 5
Setting alarm for 18:00:00 Wed 19 Jan 2022
Going to sleep at 17:57:28 Wed 19 Jan 2022
Alarm at 18:00:00 Wed 19 Jan 2022
Setting alarm for 18:05:00 Wed 19 Jan 2022
Going to sleep at 18:00:10 Wed 19 Jan 2022
Alarm at 18:05:00 Wed 19 Jan 2022
Setting alarm for 18:10:00 Wed 19 Jan 2022
Going to sleep at 18:05:10 Wed 19 Jan 2022
Alarm at 18:10:00 Wed 19 Jan 2022
Setting alarm for 18:15:00 Wed 19 Jan 2022
Going to sleep at 18:10:10 Wed 19 Jan 2022
```
```
/home/jack/Dropbox/sketchbook/temp/alarm_ex10/alarm_ex10.ino Jan 19 2022 12:42:30
Sketch start 17:42:40 Wed 19 Jan 2022
Wake interval: 5 minutes.
Using interrupt INT1
RTC interrupt pin: 3
RTC power pin: 5
Setting alarm for 17:45:00 Wed 19 Jan 2022
Going to sleep at 17:42:40 Wed 19 Jan 2022
Alarm at 17:45:00 Wed 19 Jan 2022
Setting alarm for 17:50:00 Wed 19 Jan 2022
Going to sleep at 17:45:10 Wed 19 Jan 2022
Alarm at 17:50:00 Wed 19 Jan 2022
Setting alarm for 17:55:00 Wed 19 Jan 2022
Going to sleep at 17:50:10 Wed 19 Jan 2022
Alarm at 17:55:00 Wed 19 Jan 2022
Setting alarm for 18:00:00 Wed 19 Jan 2022
Going to sleep at 17:55:10 Wed 19 Jan 2022
```

hoseinaghajari, 2022-01-19:
Unfortunately it still doesn't work for me. I will try to solve it.
why you use PIN5 for vcc?

JChristensen, 2022-01-19:
To minimize power consumption. When the MCU sleeps, it first powers off the RTC. Else the RTC will continue to draw 100 uA or so.

hoseinaghajari, 2022-01-19:
Good Idea