JChristensen/DS3232RTC #95: RTC freezes

ejmorgan55, 2022-02-08:
Running a clock based on this library example and a DS3231 RTC. I am using a multiplexer to send to different displays. After about an hour or so; the RTC sets to 0 and freezes. If I unplug and reapply power; the clock comes back on; albeit, minus the period of time it was frozen. I get no error messages. Originally running a nano and though it might be a memory problem; switched to the Mega and still have the same problems, Thought it might be wiring; however, does the same thing on a breadboard as it does on the physical clock. Any thoughts?

//PhoNixieClock
#include <DS3232RTC.h>      
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

int digit1 = 0;
int digit2 = 0;
int digit3 = 0;
int digit4 = 0;
int digit5 = 0;
int digit6 = 0;

#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 32
#define DS3231_I2C_ADDRESS 0x68
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, -1);

// Select I2C BUS
void TCA9548A(uint8_t bus){
  Wire.beginTransmission(0x70);  // TCA9548A address
  Wire.write(1 << bus);          // send byte to select bus
  Wire.endTransmission();
  Serial.print(bus);
}

void setup()
{
    Serial.begin(9600);
    RTC.begin();
    
    setSyncProvider(RTC.get);{   // the function to get the time from the RTC
      //setTime(13,49,00,14,1,2022); // Sets date-time  
    if(timeStatus() != timeSet)
        Serial.println("Unable to sync with the RTC");
    else
        Serial.println("RTC has set the system time");
    }
  // Start I2C communication with the Multiplexer
  Wire.begin();

   display.setRotation(1);

  // Init OLED display on bus number 2
  TCA9548A(2);
  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
    Serial.println(F("SSD1306 allocation failed"));
    for(;;);
  } 
  // Clear the buffer
  display.clearDisplay();

  // Init OLED display on bus number 3
  TCA9548A(3);
  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
    Serial.println(F("SSD1306 allocation failed"));
    for(;;);
  } 
  // Clear the buffer
  display.clearDisplay();

  // Init OLED display on bus number 4
  TCA9548A(4);
  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
    Serial.println(F("SSD1306 allocation failed"));
    for(;;);
  } 
  // Clear the buffer
  display.clearDisplay();

   // Init OLED display on bus number 5
  TCA9548A(5);
  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
    Serial.println(F("SSD1306 allocation failed"));
    for(;;);
  }
  // Clear the buffer
  display.clearDisplay();
  
     //Init OLED display on bus number 5
  TCA9548A(6);
  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
  Serial.println(F("SSD1306 allocation failed"));
  for(;;);
  }
  //Clear the buffer
  display.clearDisplay();
  
     // Init OLED display on bus number 5
  TCA9548A(7);
  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
  Serial.println(F("SSD1306 allocation failed"));
  for(;;);
  } 
  //Clear the buffer
  display.clearDisplay();
 
}

void loop(){

   digit1 = (hour()%10);
   digit2 = ((hour()/10)%10);
   digit3 = (minute()%10);
   digit4 = ((minute()/10)%10);
   digit5 = (second()%10);
   digit6 = ((second()/10)%10);

// Write to OLED on bus number 2
  TCA9548A(2);
  display.clearDisplay();
  display.setTextSize(5);
  display.setTextColor(WHITE);
  display.setCursor(0, 0);
  // Display static text
  display.println(digit2);
  display.display(); 

  // Write to OLED on bus number 3
  TCA9548A(3);
  display.clearDisplay();
  display.setTextSize(5);
  display.setTextColor(WHITE);
  display.setCursor(0, 0);
  // Display static text
  display.println(digit1);
  display.display(); 

  // Write to OLED on bus number 4
  TCA9548A(4);
  display.clearDisplay();
  display.setTextSize(5);
  display.setTextColor(WHITE);
  display.setCursor(0, 0);
  // Display static text
  display.println(digit4);
  display.display(); 

  // Write to OLED on bus number 5
  TCA9548A(5);
      display.clearDisplay();
      display.setTextSize(5);
      display.setTextColor(WHITE);
     display.setCursor(0, 0);
  // Display static text
    display.println(digit3);
   display.display();  

     // Write to OLED on bus number 5
 TCA9548A(6);
 display.clearDisplay();
 display.setTextSize(5);
 display.setTextColor(WHITE);
  display.setCursor(0, 0);
  // Display static text
  display.println(digit6);
  display.display();    

   // Write to OLED on bus number 5
  TCA9548A(7);
  display.clearDisplay();
  display.setTextSize(5);
  display.setTextColor(WHITE);
  display.setCursor(0, 0);
   // Display static text
  display.println(digit5);
  display.display();    

  delay(100);
}

JChristensen, 2022-02-08:
Hello @ejmorgan55,

Where did the RTC come from, i.e. where did you buy it?

JChristensen, 2022-02-15:
Nothing heard; closing issue.

closed 2022-02-15 by JChristensen