JChristensen/DS3232RTC #96: square wave seems not to work

bk227865, 2022-03-03:
enum SQWAVE_FREQS_t {
            SQWAVE_1_HZ,
            SQWAVE_1024_HZ,
            SQWAVE_4096_HZ,
            SQWAVE_8192_HZ,
            SQWAVE_NONE
        };

should this not contain values ?

JChristensen, 2022-03-04:
They do have values. If not otherwise specified, the enumerators have values 0, 1, 2, ... etc.

Is something actually not working, or is this just a question?

bk227865, 2022-03-04:
ah ok , then ... don't know if i am doing anything wrong , but i only seem to get a 1hz interrupt , whatever SQWAVE i choose (exept SQWAVE_NONE)

JChristensen, 2022-03-04:
If I run the sketch below and monitor the SQW pin with a frequency counter, all seems well. Remember that the SQW pin requires a pull-up resistor; 10K works fine.
```c++
#include <DS3232RTC.h>          // https://github.com/JChristensen/DS3232RTC

DS3232RTC myRTC;

void setup()
{
    Serial.begin(115200);
    Serial.println(F("\n" __FILE__ " " __DATE__ " " __TIME__));
    myRTC.begin();
}

void loop()
{
    Serial.println("1 Hz");
    myRTC.squareWave(DS3232RTC::SQWAVE_1_HZ);
    delay(5000);

    Serial.println("1024 Hz");
    myRTC.squareWave(DS3232RTC::SQWAVE_1024_HZ);
    delay(5000);
        
    Serial.println("4096 Hz");
    myRTC.squareWave(DS3232RTC::SQWAVE_4096_HZ);
    delay(5000);
        
    Serial.println("8192 Hz");
    myRTC.squareWave(DS3232RTC::SQWAVE_8192_HZ);
    delay(5000);
        
    Serial.println("0 Hz");
    myRTC.squareWave(DS3232RTC::SQWAVE_NONE);
    delay(5000);
}
```

bk227865, 2022-03-04:
Thank you ! 
I use pinMode(RTC_PIN, INPUT_PULLUP); but will try with a real resistor.
I guess we can close this , as it seems to be my issue.

JChristensen, 2022-03-04:
The internal pull-up resistors work fine as well. There are a lot of counterfeit and otherwise suspect parts out there. These RTCs are not inexpensive and when I see them sold for significantly less than at the top-shelf distributors, I have to wonder.
Good luck!

closed 2022-03-04 by JChristensen

seamusdemora, 2022-03-06:
> The internal pull-up resistors work fine as well. There are a lot of counterfeit and otherwise suspect parts out there. These RTCs are not inexpensive and when I see them sold for significantly less than at the top-shelf distributors, I have to wonder. Good luck!

True - The DS323X are being counterfeited on a massive scale, and the counterfeits have various defects depending on the individual counterfeiter I suppose. AFAIK, you can't buy genuine parts from Ebay, Amazon or any of the various 'direct-from-China' outlets. It's a hard world I guess.