JChristensen/DS3232RTC #97: Alarm 1 and 2

hoseinaghajari, 2022-04-01:
Hi dear @JChristensen .
This (https://github.com/JChristensen/DS3232RTC/blob/master/examples/alarm_ex6/alarm_ex6.ino) is the code which run for me correctly but I have a question. I come to think that if I just use Alarm1, there would be some problems so according to my experiences we should use both Alarms. so How could I change that code to disable correctly Alarm2 and to set Alarm1 every 5 minutes?

JChristensen, 2022-04-01:
Hello Hossein,

A combination of Example 3 and Example 6 should do the trick. Example 3 sets Alarm1 to occur at a given interval and disables Alarm2. Combine that idea with the interrupt code in Example 6.

Have a look and if you're still stuck, let me know.

Regards ... Jack

JChristensen, 2022-04-11:
Nothing heard; closing issue.

closed 2022-04-11 by JChristensen