JChristensen/DS3232RTC #98: Add example for automatically setting time

Nick-Z0, 2022-04-16:
Hi Jack,

Consider adding another example for setting and printing the time. The method in the [README.md](https://github.com/JChristensen/DS3232RTC#readme) (`myRTC.set(now());`) did not work for me since I got a date in 2066.
In your [alarm_ex1.ino](https://github.com/JChristensen/DS3232RTC/blob/master/examples/alarm_ex1/alarm_ex1.ino) you have a nice working method which also sets the compile time (which is exactly what I was looking for since I was aware of this method from other libraries) but it is not easy to find. I found it accidentally after reading almost everything else since I was not looking for an alarm. This could save new users some time.

Best regards,
Nick

JChristensen, 2022-04-16:
Hi Nick,

I'm not sure how useful setting the RTC from the compile time is, outside of an example. Compile time is a constant; once the sketch is uploaded to the MCU, it does not change. If the sketch runs for a year, then I reset the MCU, it will set the RTC to the compile time that was a year ago. Correct me if I'm wrong but that doesn't seem like something most people would want.

As for the example in the README file, it works fine for me. Give following sketch  a try; it returns `23:31:30 Fri 13 Feb 2009`
for me as expected.

```c++
#include <DS3232RTC.h>      // https://github.com/JChristensen/DS3232RTC

DS3232RTC myRTC;

void setup()
{
    Serial.begin(115200);
    myRTC.begin();
    setTime(23, 31, 30, 13, 2, 2009);   // set the system time to 23h31m30s on 13Feb2009
    myRTC.set(now());
    time_t t = myRTC.get();
    printTime(t);
}

void loop() {}

// format and print a time_t value
void printTime(time_t t)
{
    char buf[25];
    char m[4];    // temporary storage for month string (DateStrings.cpp uses shared buffer)
    strcpy(m, monthShortStr(month(t)));
    sprintf(buf, "%.2d:%.2d:%.2d %s %.2d %s %d",
        hour(t), minute(t), second(t), dayShortStr(weekday(t)), day(t), m, year(t));
    Serial.println(buf);
}
```

Nick-Z0, 2022-04-16:
I would use compile time as setting the time and then commenting the line out in the sketch that would be actually running. When the RTC battery runs out and you replace it, you just have to uncomment a line and do it again (you could also let an unacquainted person do this since it is simple).

I find hardcoding the date-time and clicking upload at a specific moment cumbersome and consider this an easier method. 
(Btw the fudge factor works great, other libraries don't have such a function so their compilation time is off by a couple of seconds)

About the example in the README file, I see that I used it wrong. I thought that `myRTC.set(now());` was a standalone command to get the computer time.

JChristensen, 2022-04-16:
Since that requires uploading the sketch two more times I might just use the SetSerial example, which would probably be more accurate.

closed 2022-04-16 by JChristensen