JChristensen/DS3232RTC #99: error: 'class DS3232RTC' has no member named 'now'

australopitheque, 2022-04-20:
Good morning,
I know that this is not really an error but would it be possible to add the subclass "`Now()`" in DS3231RTC.
It would avoid adding a library that could add errors since this one works perfectly.

example that does not pass:

`DateTime now = myRTC.now();`

JChristensen, 2022-04-20:
I'm not familiar with the `DateTime` data type. Would you provide a reference please?

australopitheque, 2022-04-21:
DateTime et now() est dans la librairie RTClib.h

`
DateTime DS3231::now() {
    WIRE.beginTransmission(DS3231_ADDRESS) ;
    WIRE.write(DS3231_TIME_ADDR) ;
    WIRE.endTransmission() ;
    WIRE.requestFrom(DS3231_ADDRESS, 7) ;
    uint8_t ss = bcd2bin(WIRE.read() & 0x7F) ;
    uint8_t mm = bcd2bin(WIRE.read()) ;
    uint8_t hh = bcd2bin(WIRE.read()) ;
    WIRE.read() ;
    uint8_t d = bcd2bin(WIRE.read()) ;
    uint8_t cen = WIRE.read() ;
    uint8_t m = bcd2bin(cen & 0x1F) ;
    uint16_t y = bcd2bin(WIRE.read()) ;
    si ((cen & 0x80) >> 7 == 1) {
        y += 2000 ;
    }
    sinon {
        y += 1900 ;
    }
    return DateTime(y, m, d, hh, mm, ss) ;
}
`
and
`
// Simple general-purpose date/time class (no TZ / DST / leap second handling!)
class DateTime {
public:
    char* format(char* ret);
    char* tostr(char* charr);
    DateTime(uint32_t t = 0);
    DateTime(uint16_t year, uint8_t month, uint8_t day,
             uint8_t hour = 0, uint8_t min = 0, uint8_t sec = 0);
    DateTime(const char* date, const char* time);
    DateTime(const __FlashStringHelper* date, const __FlashStringHelper* time);
    DateTime(const char* sdate); // Do we really need this?
    uint16_t year() const { return 2000 + yOff; }
    uint8_t month() const { return m; }
    uint8_t day() const { return d; }
    uint8_t hour() const { return hh; }
    uint8_t minute() const { return mm; }
    uint8_t second() const { return ss; }
    uint8_t dayOfWeek() const;
    void SetTime(const char* time);
    void SetDate(const char* date);
    void setyear(uint16_t year) { yOff = year - (year >= 2000 ? 2000 : 0); }
    void setmonth(uint8_t month) { m = month; }
    void setday(uint8_t day) { d = day; }
    void sethour(uint8_t hour) { hh = hour % 24; }
    void setminute(uint8_t minute) { mm = minute % 60; }
    void setsecond(uint8_t second) { ss = second % 60; }
    // 32-bit UNIX timestamp
    // An uint32_t should be able to store up to 2106,
    // which is beyond most chip's upper bound 2099
    void setunixtime(uint32_t t);
    uint32_t unixtime() const;

    bool operator==(const DateTime& date) const;
    bool operator==(const char* sdate) const;
    bool operator!=(const DateTime& date) const;
    bool operator!=(const char* sdate) const;
    bool operator<(const DateTime& date) const;
    bool operator>(const DateTime& date) const;
    bool operator<=(const DateTime& date) const;
    bool operator>=(const DateTime& date) const;

    DateTime operator+(uint32_t t) const;
    DateTime operator+(const TimeDelta& delta) const;
    DateTime operator-(uint32_t t) const;
    DateTime operator-(const TimeDelta& delta) const;
    TimeDelta operator-(const DateTime& date) const;

    DateTime& operator+=(uint32_t t);
    DateTime& operator+=(const TimeDelta& delta);
    DateTime& operator-=(uint32_t t);
    DateTime& operator-=(const TimeDelta& delta);

protected:
    uint8_t yOff, m, d, hh, mm, ss;
};
`

JChristensen, 2022-04-21:
There are at least a couple libraries with that name. I can't guess which it is. Please provide a link.

australopitheque, 2022-04-21:
https://github.com/NeiroNx/RTCLib?utm_source=platformio&utm_medium=piohome

australopitheque, 2022-04-21:
adafruit too
https://github.com/adafruit/RTClib/blob/master/src/RTC_DS3231.cpp

australopitheque, 2022-04-21:
I use the adafruit one

JChristensen, 2022-04-21:
If you're just wanting Unix time, then that should work. Example sketch below.
If you're wanting the `DateTime` class replicated in my library, then I don't think that will happen.
If you're wanting something else, then let me know what it is.
```c++
#include <DS3232RTC.h>      // https://github.com/JChristensen/DS3232RTC

DS3232RTC myRTC;

void setup()
{
    Serial.begin(115200);
    myRTC.begin();
}

void loop()
{
    time_t now = myRTC.get();
    Serial.print(year(now), DEC);
    Serial.print('/');
    Serial.print(month(now), DEC);
    Serial.print('/');
    Serial.print(day(now), DEC);
    Serial.print(" (");
    Serial.print(dayShortStr(weekday(now)));
    Serial.print(") ");
    Serial.print(hour(now), DEC);
    Serial.print(':');
    Serial.print(minute(now), DEC);
    Serial.print(':');
    Serial.print(second(now), DEC);
    Serial.println();
    delay(3000);
}
```

australopitheque, 2022-04-21:
je vous remercie beaucoup

closed 2022-04-21 by australopitheque

JChristensen, 2022-04-21:
Glad to have helped. Best regards from the USA!