JChristensen/DS3232RTC #100: How to set an alarm every two days?

anrada73, 2022-06-09:
Nice library, thanks. How is it possible to easily turn on the alarm at a given time in 1 or 2 days?

JChristensen, 2022-06-09:
The technique would be the same as the example sketch [alarm_ex3](https://github.com/JChristensen/DS3232RTC/blob/master/examples/alarm_ex3/alarm_ex3.ino), except when [setting the alarm](https://github.com/JChristensen/DS3232RTC/blob/ef57d34517dec83029288e771add6004b46fba04/examples/alarm_ex3/alarm_ex3.ino#L54), use ALM1_MATCH_DATE instead of ALM1_MATCH_HOURS.

closed 2022-06-09 by JChristensen

anrada73, 2022-06-09:
You mean ALARM_INTERVAL ?  If in a day, then set to 86.400 seconds?

JChristensen, 2022-06-09:
I mean:
```c++
myRTC.setAlarm(DS3232RTC::ALM1_MATCH_DATE, second, minute, hour, date);
```