JChristensen/DS3232RTC #101: myRTC.set(now()) not working Anymore

MacMattia, 2022-11-04:
Hello Jack,

I was using this library with success but now (when i was updating my code) the function myRTC.set(now()) stopped working right for me. It always sets actual time to midnight if 01Jan2066. I tried with different examples to understand the issue but I don't know why. With the SetSerial.ino it reads correctly the date and time from the PC but then it still sets it to midnight 01Jan2066.
The RTC D3231 module works right because it sets the right time with other libraries. Tested on different boards (one original Arduino Uno and one clone) but the issue is the same.
Thank you very much in advance for the support

Best regards

Mattia

JChristensen, 2022-11-04:
Hello Mattia,

Are you saying that it worked correctly "before" but now does not work correctly? What happens if you revert your code to the previous version that worked?

What microcontroller are you using?

Regards ... Jack

MacMattia, 2022-11-04:
Hello Jack,

I wrote the code a couple of weeks ago, now I had to do a little change in a variable that was not related to the RTC. I uploaded the code and saw this issue with date and time. So I tried also other codes of the basic examples in your library and the "now" time is always 00:00 01Jan2066. With other libraries (like RTCLib) the time setup is working correctly.
I'm using and Arduino UNO Rev3 and the Elegoo clone of the same board, same issue.

Thank you for the prompt feedback

Mattia

JChristensen, 2022-11-04:
I can only think there is a hardware issue if the example sketches do not produce the expected results. For example, running the SetSerial example, unchanged, on an Uno or clone.

JChristensen, 2022-11-04:
I had a board handy, so I tested the SetSerial example using both IDE version 1.8.19 and version 2.0.1. I didn't expect anything different, but it works fine with both.

MacMattia, 2022-11-04:
I tried several times with SetSerial example and with my code, the present hour became always 00:00 31Dec2095... Tried with RTCLib and it worked well as usual. Then I tried again with the DS3232RTC library tinkering a bit, I set the hour manually and it worked, back again to the "myRTC.set(now())" and now looks like it's working again, both in my code and in the examples. Didn't touch anything on the hardware side.

To be honest also in the past i had this issue, but that time the "myRTC.set(now())" was drifting from the PC time of 1 hour and some minutes. Also that time it worked again the day later without changing anything.

Still a mystery

Thanks again for the support

JChristensen, 2022-11-04:
`now()` is the current time from the [Time library](https://github.com/PaulStoffregen/Time). If it has not been set first, then setting the RTC via `myRTC.set(now())` may not yield the desired results.

The Time library is essentially a software RTC that runs on the microcontroller. So we have two independent clocks running (the other being the RTC.) By default, the Time library syncs itself with the RTC every five minutes.

Not saying that is the issue, but just FYI in case you were not aware. Also see the [rtc_interrupt](https://github.com/JChristensen/DS3232RTC/blob/master/examples/rtc_interrupt/rtc_interrupt.ino) example which bypasses the Time library's RTC and keeps the MCU's time exactly in sync with the RTC.

closed 2022-11-04 by JChristensen